EdgeTX/edgetx #3039: TX16S on Edgetx 2.8.0 randomly shutsdown and boots into EM when in Joystick mode through USB connection

Bottledg, 2023-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I updated my TX16S (original) to EdgeTX 2.8.0 (firmware and SD card). Everything went fine and I had no issues using the radio except for when playing simulators (Liftoff, Velocidrone, DRL) with a USB cable (the original that came with the radio). The radio will randomly shutdown and when I turn it back on it is in emergency mode. Sometimes it happens when the games load.  Other times it will happen within a few minutes of playing.
I tried calibrating the radio in the windows 10 "Setup USB game controller" program and found that it also caused the radio to shutdown and reboot into emergency mode.

### Expected Behavior

Radio shouldn't crash when in Joystick mode and connected to USB.

### Steps To Reproduce

1. Connect radio to computer using a USB cable and select Joystick mode.
2. Launch a simulator such as Liftoff, Velocidone, or DRL.
3. Radio will either shut down and on simulator startup or within the first 5-10 minutes of playing.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

I tried a new cable. The issue continued to happen. 
I tried a new SD card (formatted first). The issue continues to happen.
I reflashed the firmware. The issue continues to happen.
I made sure that my battery was full (Radiomaster 5000mah li-ion pack). The issue continues to happen.
Switched to using a 2s 1300mah Gens Ace LiPO. The issue continues to happen.
I redownloaded and reflashed the firmware. The issue continues to happen.
Deleted all radio and model settings from the SD card and started from scratch. The issue continues to happen.
I reloaded my EdgeTX 2.7.1 backup on my original SD card, battery and USB cable. The issue disappeared.
Loaded EdgeTX 2.8.0 again and the issue happens again.

Can provide a copy of my SD card contents if you want to test.

Bottledg, 2023-03-03:
Tested 2.8.1. The issue still persists even when I try all of the above tests.

Bottledg, 2023-03-12:
Flashed EdgeTX "Flying Dutchman" v2.8.0-RC4 to my radio and the issue was gone. Flashed back to EdgeTX "Flying Dutchman" v2.8.0 and the issue reappeared. Looks like whatever changes were made between v2.8.0-RC4 and v2.8.0 are the cause of the problem.

Bottledg, 2023-03-19:
I've narrowed it down to the issue happening when I am using the BATT Analog widget on 2.8.0 and 2.8.1. When I remove the widget everything works fine.

gagarinlg, 2023-03-19:
Wow, that's strange.
You you please zip and upload theradio and model folders?

Bottledg, 2023-03-19:
[RADIO_and_MODELS.zip](https://github.com/EdgeTX/edgetx/files/11012421/RADIO_and_MODELS.zip)
It should be noted that I am using a first generation Radiomaster TX16s that has been upgraded with AG01 gimbals. Someone in the discord group tested this with a TX16S MK2 and said that they were not able to replicate the issue. No one has stepped forward to test with a first generation TX16S.

gagarinlg, 2023-03-19:
There are no HW differences between those two versions that make a difference for such an error

philmoz, 2023-03-19:
I loaded the full SD card copy that Bottledg linked on discord onto my TX16S Mk2 (with AG01 gimbals) and flashed it with 2.8.1.

I was unable to reproduce the crash. Tested with RealFlight 9.5, Aerofly RC8 and the ETX simulator.

@Bottledg have you done any other mods to the radio other than the AG01 gimbals?

Bottledg, 2023-03-19:
I replaced the radio handle but the gimbals are the only things I have changed that deal with the electronics.

Bottledg, 2023-04-02:
Today I tried EdgeTX 2.8.2. My radio didn't crash and go into Emergency mode but my BATT Analog widget was replaced my a memory error.

[image]

Could this be what was causing the crashes?

pfeerick, 2023-04-03:
If you have a spare SD card (so you don't mess up your current models and settings), it would be interesting to see if a nightly (for 2.9) also has issues, as there have been some changes to the joystick code... *maybe* something changed there that fixed this, or perhaps at least change the behaviour and help narrow down the culprit?

Bottledg, 2023-04-04:
The crashes into emergency mode happen more frequently and within a minute or two of being connected in joystick mode using the 2.9.0 nightly build. Whatever this problem is started with 2.8.0 and it looks like it will get worse with 2.9.0. If I don't use widgets it takes longer to crash. When I use widgets I usually get crashes within 4-8 minutes on 2.8.x. 2.7.1 was the last version that I was able to use on my radio without any issues. I know that the USB boards are different for the TX16S MK2 radios that the original TX16S which is what I have. People on Discord have tried to help me but they aren't testing with the same version radio that I am. Then again, maybe it's just my radio that is the problem. Not sure what else to try other than stick to 2.7.1 .

gagarinlg, 2023-04-04:
I do not have this issue with 2.8.1
I tested with a TX16S MKI and MKII

millerjef, 2023-04-09:
I've routinely experienced the same issue with both a TX16S MK1 and MK2 connected via USB-C to USB-C and USB-C to USB-A on Windows 10 and Windows 11 in the Steam version of RealFlight Evolution.

wojtek3, 2023-05-10:
I've experienced the same issue with USB Joystick on tx16s (mark 1). Moreover i noticed that it also happens when radio is connected as USB Storage (it crashes randomly - sometimes after 10 seconds, sometimes after 5 minutes or more). I flashed EdgeTX 2.7.1 and issue disappeared.

philmoz, 2023-05-10:
> I've experienced the same issue with USB Joystick on tx16s (mark 1). Moreover i noticed that it also happens when radio is connected as USB Storage (it crashes randomly - sometimes after 10 seconds, sometimes after 5 minutes or more). I flashed EdgeTX 2.7.1 and issue disappeared.

Do you have any widgets loaded on the screen when using the joystick? If so which ones?

wojtek3, 2023-05-11:
> Do you have any widgets loaded on the screen when using the joystick? If so which ones?

I was using outputs widget and value widget (to show battery voltage), but I also experienced this problem when using a clean model without widgets and I don't feel any change when using clean model. I tried to flash 2.8.0 and 2.8.3 firmware, but it doesn't change anything.
Is there any way to check what is causing this problem? I have some some electronic stuff like uart interfaces, st-link programmer and more, so if there is any way to debug I can help.

gagarinlg, 2023-09-20:
> I was using outputs widget and value widget (to show battery voltage), but I also experienced this problem when using a clean model without widgets and I don't feel any change when using clean [...]

It is a bit late for a reply, but still
since you have the issue, can you compile the firmware and run it from a debugger. It should stop then when the radio would reboot into EM, so that you might be able to see where the EM was triggered

closed 2026-09-21 by github-actions[bot]