EdgeTX/edgetx #3040: LibSimulator won't link with QT under Apple M1 processor

nuttybrewer, 2023-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

Followed instructions from wiki, updated QT from 5.12.9 to 5.12.12. note, firmware compiles fine, this seems to be a QT issue. Not sure QT5 can work native with M1 silicon as per https://www.qt.io/blog/qt-on-apple-silicon (I tried to define the silicon as per blog entry.

Searches across the googles seems to indicate that QT5 needs to be compiled as arm64 as it is delivered by QT as x86_64.

Tried with brew install qt@5 with the same result.

### Expected Behavior

QT5 links with arm64 library and not x86_64 library.

Only solution may be to upgrade to QT6.2+

### Steps To Reproduce

Followed instructions from wiki, updated QT from 5.12.9 to 5.12.12. note, firmware compiles fine, this seems to be a QT issue.
rm -rf edgetx
git clone edgetx
mkdir -p edgetx/build
cd edgetx/build
cmake -DPCB=X10 -DPCBREV=TX16S -DCMAKE_PREFIX_PATH=$QTDIR -DGCC_ARM_PATH=/Applications/ARM/bin/ -DPYTHON_EXECUTABLE=/Users/<user>/venv/edgetx/bin/python3 -DCMAKE_APPLE_SILICON_PROCESSOR=arm ..
make -j8 firmware ; #SUCCEEDS and flashes to radio!
make libsimulator ; #Fails with message below
....
[100%] Linking CXX shared library ../../../../libedgetx-tx16s-simulator.dylib
ld: warning: ignoring file /Users/<user>/venv/edgetx/5.12.12/clang_64/lib/QtCore.framework/QtCore, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:

### Version

2.8.0

### Transmitter

Radiomaster Zorro

pfeerick, 2023-01-15:
QT5 issues AFAIK... as mentioned in that doc, Qt 5.15.4 is the version needed for those patches, and even then, they are only talking about the commercial version of QT5 - this is why we need to port stuff over to QT6, which is no small challenge.

nuttybrewer, 2023-01-15:
Agreed with your conclusion.

We can recompile QT https://wiki.qt.io/Building_Qt_5_from_Git#Getting_the_source_code. I will try this and post back findings here once I'm done.

nuttybrewer, 2023-01-15:
Success! I don't know why QT isn't just redistribution the arm64 distribution, but nonetheless here are the extra steps to be added to the EdgeTX MacOS dev wiki to compile QT...
```
# Set up CLANG environment
brew install llvm
export PATH="/opt/homebrew/opt/llvm/bin:$PATH"

# Create install directory for QT (to be used as QTDIR in the EdgeTX cmake later)
sudo mkdir -p /opt/qt5
sudo chmod <i>user:group</i> /opt/qt5

# Move to your git environment
git clone https://code.qt.io/qt/qt5.git
cd qt5
git tag # List all tags, see which is the latest v5.x release
git checkout v5.15.8-lts-lgpl
git submodule update --init --recursive

# Make a shadow build directory
mkdir qt5-build && cd qt5-build

# Configure QT build
../configure -opensource -nomake examples -nomake tests -confirm-license -prefix /opt/qt5 -skip qtwebengine

# Make
make -j $(nproc) # or # of CPU cores instead of $(nproc)

# Install
make install
```
Keep building edgetx as required (define QTDIR=/opt/qt5/)

If somebody could confirm this and then go ahead and close this issue?

pfeerick, 2023-01-15:
@raphaelcoeffic You might be interested in this ;)

nuttybrewer, 2023-01-15:
[image]

There's still some problems compiling other radios, but this one works fine. I need to switch my branch to a stable one to test further.

philmoz, 2023-12-13:
Can probably be closed. Current build of 2.9 and 2.10 work on M2 Mac with Ventura.

3djc, 2023-12-13:
and on M2 Mac with Sonoma

closed 2023-12-13 by pfeerick

raphaelcoeffic, 2023-12-13:
> Can probably be closed. Current build of 2.9 and 2.10 work on M2 Mac with Ventura.

You have a M2 now? Am I the only one with M1 Pro on this earth? 😭