EdgeTX/edgetx #3047: Companion Telemetry Simulator should pull telemetry sensor list from model data

Dweinbach, 2023-01-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The companion telemetry simulator seems to have only a hardcoded list of sensors, which do not match my models very well. This makes it basically useless.

### Describe the solution you'd like

The Companion Telemetry Simulator should pull its telemetry sensor list from model data. When launching the telemetry simulator, it should do a fresh pull of telemetry values and ADD them to the simulator. it would be good to identify the sensors that are in the latest pull with a different background color or some other indicator. 
When exporting or importing, the list should be the superset of all known (ever pulled + default for edgeTX) values. it is not necessary for ETX to automatically prune this import/export/table. It is good to have all possible values from a user's ecosystem, IMO.

ParkerEde, 2023-02-03:
The problem is that when the simulator is started, the sensors are not read in at first. If you then press the button "Reload all Radio Data"
[image]
while the simulator is running, the sensors are read in according to the model and work. 
BTW the GPS data look very strange, a revision might also be necessary here
[image]

rotorman, 2024-08-02:
https://edgetx.org/contest_crsf_telem_simulation/