EdgeTX/edgetx #3048: Edge parameter change after edit and exit.

Kevltan, 2023-01-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If in edge command you edit the delay time and exit with for instance [1.5:<<] the second parameter reverts to "39.5" [1.5:39.5]. You have to reselect the [<<] parameter before exiting or come back in and change back afterwards. It does work but it is strange that you have to revisit this and check that it has remained as set.

### Expected Behavior

Change the parameter selected in edge and expect no change to second parameter when exiting.

### Steps To Reproduce

in 2.9.0 nighltly but expect is also in be 2.8.0

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

14-01-23
[image]
Before edit
[image]
After editing and exiting "<<" changes to "39.5" should have remained "<<" and believe should not have to reselect or revisit and change after exit.

philmoz, 2023-01-16:
I've created a PR to fix this - it might take a little while to get merged into the nightly build.

Kevltan, 2023-01-16:
Thankyou 😊

Kevltan, 2023-01-17:
Made edit to firmware as indicated rebuilt and tested and now works fine. Thanks Phil 👍

PR #3049 (2023-01-17): fix(ui) - Fix update of V3 when V2 is modified whilst editing a logical switch.
Fixes #3048 (#3035)

Summary of changes:

Fix to correctly update the second time value (V3) for 'Edge' switches when V2 is updated, and also display the new V3 value immediately.
closed 2023-01-17 by pfeerick