EdgeTX/edgetx #3054: Wireless trainer connection using internal MPM intermitent when external ELRS module is turned on

Fsarmento, 2023-01-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I want to use a FlySky fs-i6x transmitter as a Student connected to my Radiomaster TX16S on the internal MPM using the FlySky AFHDS 2A RX protocol. The TX16S is connected to the model with an external ELRS module.

Here are the steps I did:
1) Turn OFF the External module
2) Set the Trainer mode (button to activate Trainer mode on Special Functions and Trainer mode to Master/Multi)
3) Bind the FS-i6x radio to my TX16s

At this stage, the Trainer model works fine. I can confirm on the Channel Monitor that I have control if the Trainer button is disabled, and the FS-i6x has control if the Trainer button is enabled.

The strange behavior happens when I turn on the external ELRS module. As soon it is turned on, the Student/Trainer connection becomes intermittent, jumping from connected to Trainer connection lost every second. 

I am posting a video of the behavior for a better understanding.

https://user-images.githubusercontent.com/18128610/212780579-65ea1cc2-d674-443a-9ebb-840ab4c9624e.mp4

### Expected Behavior

The connection between the fs-i6x and the TX16S should continue to be stable even when the external module is ON.

### Steps To Reproduce

1) Turn OFF the External module
2) Set the Trainer mode (button to activate Trainer mode on Special Functions and Trainer mode to Master/Multi)
3) Bind the FS-i6x radio to my TX16s
4) Confirm Student/Trainer behaviour works as expected on Channel Monitor
5) Turn on the external module on CRSF (ExpressLRS)
6) Strange behavior happens - Connection between Student and Master radio becomes intermittent.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-01-17:
What packet rate are you using (and possibly the baud rate)... this advisory applied as of 2.7 (probably has always been the case, but the issue was discovered around that time):
> Wireless Model Trainer and CRSF/ELRS - if using the Multi Protocol Module (MPM) as a wireless trainer in conjunction with CRSF/ELRS, do not go above 250 Hz, otherwise the MPM wireless trainer will be dropped. 250 Hz or below is stable.

Fsarmento, 2023-01-18:
Did some tests and concluded what you stated. For Packet Rates of 250Hz or lower, the connection between the student and Master radios is stable.  From Packet Rates of 333Hz or higher the trainer connection drops.
Regarding the baud rate, I didn't see any implication on Trainer connection drop. I guess that to use the Full res 8ch I will have to go for the 100Hz packet rate. Thanks for the info.

pfeerick, 2023-01-18:
Yes, at the end of the day it boils down to the packet rate, as higher baud rate only *allow* for higher packet rates. This should change in the future as the limitation is actually the packet rate of the MPM, and we are working on decoupling that so that the internal and external modules can run at their native packet rates, rather than be limited to the rate of the MPM when it's enabled.

mha1, 2023-01-19:
@pfeerick I'd appreciate you reviewing this too: https://github.com/EdgeTX/edgetx/pull/2920

mha1, 2023-01-19:
2.8 and current main force the external ELRS module to the MPM's update period of 7ms (143 Hz) if both modules are enabled. Setting the ELRS's packet rate to rates higher than 100Hz will not generate the desired packet rates but will be limited to 143Hz. You can check yourself using the ELRS LUA script (top right corner). Even if there are no obvious faults at 250Hz it will still only have a packet rate of 143Hz. There is really no point in setting the packet rate higher than 100Hz when the MPM is also enabled.

This PR is an attempt to allow higher packet rates but is still work in progress: https://github.com/EdgeTX/edgetx/pull/2920