EdgeTX/edgetx #3056: New external module protocol: telemetry mirror

r1000ru, 2023-01-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Most transmitters do not have a built-in BLE or the ability to install it. At the same time, users of transmitters with built-in ELRS often do not use external modules, but would like to receive telemetry on a phone or antenna tracker. 

### Describe the solution you'd like

I propose to make it possible to connect the BLE module to the SmartPort pin of the external module connector. One transistor inverter + HC-05 or ESP32 (can work with inverted uart) could transmit telemetry

### Describe alternatives you've considered

1. Buy another tranmitter
2. Interception packets beetween Int module TX and main board RX (it is work)

pfeerick, 2023-01-18:
So really you need better documentation of this, or how to access it on given transmitters?

[image]

r1000ru, 2023-01-18:
I don't use TX16/TX12mk2/Zorro, which have UART AUX. My T-Pro doesn't have it and only the SmartPort on the external connector remains.

brainbubblersbest, 2023-01-19:
The Horus Radios dont have external connectors.
Using the s.port connector would be an easy alternative here. Olliw42 does that for his mav-link project.

The x10express dont allow to change the aux Port used in the hf bay as well for a telemetry mirror usage. And here it is a two wire serial from factory.
X10 and x12 allow that.

roroid, 2023-01-30:
> So really you need better documentation of this, or how to access it on given transmitters?
> 
> [image]

Hi, sorry for barging in. I was looking for something similar, and google brought me here.

Can someone tell me where the physical AUX1 is on the Zorro radio transmitter?
I have to agree with the guy that initiated this issue. It would have been much easier to use the s.port on the external transmitter for connecting a Bluetooth module.

Thanks.

rotorman, 2023-01-30:
@roroid See the R (=RX), T (=TX) and GND pads on Zorro mainboard:
[image]

roroid, 2023-01-30:
> @roroid See the R (=RX), T (=TX) and GND pads on Zorro mainboard:
> [image]
[image]
[image]

Man you are my only hero!!! Thank you! 
Do you know if the corect baud rate is 57600 on that port?

rotorman, 2023-01-30:
Depends on the application. Try 115200 8N1.

druckgott, 2023-06-10:
I use a Taranix x-lite pro and Expresslrs, where can I find this. I only find in the second page unter Trainer --> Off/Master/Slave, but I can not connect to it, I get an error. Where can I select telemetry Mirror with the integrated bluetooth modul?

3djc, 2023-06-10:
Taranix x-lite pro doesn't have aux port

druckgott, 2023-06-10:
But an internal Bluetooth Modul, I used it with frsky and opentx. But now with edge tx and express lrs did not work

druckgott, 2023-06-11:
I found the solution. I have to switch off trainer mode then it works.

therooftopprinz, 2024-06-01:
is it possible to mirror it to usb when connected as uart

pfeerick, 2024-06-01:
> is it possible to mirror it to usb when connected as uart

Do you mean the Telem Mirror option of USB-VCP (when USB-VCP is option for your radio)?

https://manual.edgetx.org/color-radios/radio-settings/hardware#serial-port
https://manual.edgetx.org/bw-radios/radio-settings/hardware#additional-hardware-configuration-options

rotorman, 2025-09-01:
[image]