EdgeTX/edgetx #3057: Input editor may show the wrong curve for inputs with multiple lines

philmoz, 2023-01-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Where an input has multiple lines with different weight, offset, expo or custom curve, the input editor displays the curve for the active line (based on switch selection), not the line that was selected for editing.

Editing the input line values will not update the curve preview in this case.

### Expected Behavior

The input editor should display the curve preview for the line being edited not the active input line.

### Steps To Reproduce

1. Create an input with multiple lines controlled by a switch.
2. Set input line values so the curves are obviously different.
3. Edit the active line - note the curve preview is correct and updates when values are changed
4. Edit an inactive line - note the curve preview will be from the active line and does not update

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

eshifri, 2023-01-17:
It was always this way starting from old OTX versions.
I always considered it to be a "strange feature" rather than a bug. :-)
Actually it is kind of convenient albeit not too logical. :-)

philmoz, 2023-01-17:
I'm not sure how seeing the wrong curve and not seeing it update as you edit the input values is convenient; but if that's the expected behaviour then I'll leave it alone for now.

eshifri, 2023-01-17:
It reacts to the switch and it is convenient to curves for all switch positions.
Anyway, it is one man opinion only. 
Maybe other users/developers will also consider it a bug.
Maybe keep it open for other people to comment.

eshifri, 2023-01-18:
Yes, this is exactly what I meant: it is convenient to see all setups, attached to the switch.
Maybe we should switch between mixer lines, based on the switch position? But on the other hand, in this case selection of the line for editing does not mean anything.... 
I do not have good ideas.

pfeerick, 2023-01-18:
It's almost like a toggle is needed to show active input vs input being edited.

philmoz, 2023-01-18:
> It's almost like a toggle is needed to show active input vs input being edited.

You read my mind :)

[image]

closed 2026-03-28 by philmoz