EdgeTX/edgetx #3066: Flysky Gimbal Support for RM Boxer

richardclli, 2023-01-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Same as TX16S, will work out the driver with auto detection.

### Describe the solution you'd like

Added the use of flysky gimbal driver in boxer definitions.

richardclli, 2023-01-20:
PA0: Stick_RH -> UART4_TX
PA1: Stick_RV -> UART4_RX

raphaelcoeffic, 2023-02-11:
> good day, sorry for writing in court, but I need help, I bought a Boxer radiomaster and it doesn't turn on, I charged the batteries, but it still doesn't work, the batteries are 18650 Help! please because the radiomaster wrote to me to unlock or buy other batteries, alas, since I'm a complete beginner, I don't know how to even unlock it

This is not the place to discuss your problem. We are discussing code changes here. Please contact us on Discord, or directly the Radiomaster support.

AntonZatvarskyy, 2023-02-11:
unfortunately, I have been communicating with support for a couple of days,
and I would be happy to contact you on Discord, but my English is bad, but
you can try

AntonZatvarskyy, 2023-02-11:
I can speak in Russian or Ukrainian or we will correspond in discord

rotorman, 2023-02-11:
> PA0: Stick_RH -> UART4_TX
> PA1: Stick_RV -> UART4_RX
> Similar to horus flysky gimbal support.

RV and RH are swapped above. Correct for RadioMaster Boxer is:
PA0/UART4_TX: Stick RV in analog mode
PA1/UART4_RX: Stick RH in analog mode

Whereas on T16/TX16S/T18 (á la "Horus"), the UART4 TX/RX (PA0/PA1) mapping is as:
PA0/UART4_TX: Stick LH in analog mode
PA1/UART4_RX: Stick LV in analog mode

PR #3139 (2023-02-26): Flysky gimbal support for Boxer

Fixes #3066

Summary of changes:
closed 2023-02-26 by rotorman