EdgeTX/edgetx #3069: CYC3 output does not work correctly

Tribe58, 2023-01-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When setting up a flybar heli with 120 swash, channel 6 servo (CYC3) movement does not match simulator of EdgeTX companion.  Channel 6 servo follows throttle curve when it is referenced to pitch curve.  See attached doc for setup.

[Heli Setup Issue.docx](https://github.com/EdgeTX/edgetx/files/10463110/Heli.Setup.Issue.docx)

### Expected Behavior

Channel 6 output on radio should match EdgeTX companion simulator.

### Steps To Reproduce

Changed the pitch curve with no change to channel 6 servo output.  Any change in throttle curve resulted in changed behavior of channel 6 servo.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

jrwieland, 2023-01-20:
Your use of the cyclic (sources and mixes) is off.  I'm assuming you followed a youtube to get your settings.  This was a very old work around prior to the heli tab being introduced.

There is going to be a Heli wizard added to EdgeTX soon.  Until then you can download the script from here
https://github.com/jrwieland/edgetx-sdcard-wizard

I have created a model for you to use based on your curves.  Make sure you have the esc disconnected, the curves you stipulate have the motor running when the throttle is at -100 [CV1: Thr  100, -100, 0, 100, -100 (set this way to verify fault)] I know you put a note that it was for testing but that is a scary curve.

It demonstrates the proper use of the heli tab sources as well as how the mixes are set.  The model was created using a TX16 with TAER to match what is listed in your document.

[450SE-CCLP.zip](https://github.com/EdgeTX/edgetx/files/10468784/450SE-CCLP.zip)

Tribe58, 2023-01-21:
First off, THANK YOU for your help.

I downloaded your model, loaded it to my radio, and bound the helicopter receiver.  I am experiencing the same results.  I realize that my throttle curve in the example given was scary, but my intent was for the servo in question to mimic what parameters were set in the curve and never to actually run the helicopter that way.  Regardless, I should see no movement of any swashplate servos that follows the throttle curve.  They should all follow the pitch curve, or cyclic inputs. 

I have a Spektrum AR6200 receiver (Throttle, Aileron, Elevator, Rudder, Gear, Aux 1).  The swashplate is a Y swashplate.  Looking from the tail, the forward left servo is connected to Ail, the forward right servo is connected to Aux 1, the aft center servo is connected to Elev.  

The forward left servo, which is connected to the AIL port of the receiver is the one that follows the throttle curve.  The other two work correctly.  The channel monitor on the radio shows the correct movement, but that one particular servo on the helicopter does not follow that command.  It's actually channel 2 (Ail) that's not working correctly and not channel 6 as previously mentioned.  I swapped out the receiver with a different one and got the same results, so that isn't the issue.

This radio is brand new.  I updated the firmware to 2.8.0 and enabled the Bluetooth and Heli tab when it was compiled.  Nothing else was changed.  I am starting to believe that I might have gotten a bad radio and I need to figure it out in the next few days before the return window is closed.

jrwieland, 2023-01-21:
The wiring is correct. 

The curve the swash is following is a straight -100 to +100 not the "scary" one.

Moving just the elevator all 3 should move up/down in sync. 

Since this isn't a  bug.  Please contact me directly to get this sorted out.

John.r.wieland@gmail.com