EdgeTX/edgetx #3079: Companion Simulator joystick missing channel 5 and 7

psychotik2k3, 2023-01-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

like explained in issue [#2924](https://github.com/EdgeTX/edgetx/issues/2924) (but the title is not very clear)
if i want to use my TX16S as input in the simulator, all channels works fine except channel 5 and 7 that are not moving at all.

The problem comes from the companion as when i look on the windows joystick properties the channels moves correctly.

here is a screenshot of the problem, channel5 = "axe de rotation Y" on windows joystick properties
[image]

### Expected Behavior

Channel 5 and channel 7 should move accroding to the channel on the joystick on windows.

### Steps To Reproduce

conenct your transmitter as usb joystick and with a simple program where you mapp all sticks, then all sliders, then all pots as channels directly.
on companion settings go to settings=> simulator settings, joystick select your radio check the active check ond go to calibrate so monitor the channels.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-01-24:
Duplicate of #2924

philmoz, 2023-01-27:
This may be fixed with the change to SDL2 (PR #3085)

I tried a Windows build with this PR and all the joystick inputs appear to work.

gagarinlg, 2023-09-18:
@psychotik2k3 can you please re-test with 2.9

closed 2023-10-19 by github-actions[bot]