EdgeTX/edgetx #3083: getSwitchIndex("L01") does return nil if switch is not defined

ChrisOhara57, 2023-01-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I write scripts in LUA. Usually I look for magic numbers, in this case it is the switch index of Logical Switch L01.
I use this number as reference by adding up values to get the index position of the other logical switches.

Changing my existing code to use getSwitchIndex("L01")  got  me into trouble.
In fact I do get 36 only if the logical switch L01 is really defined. Otherwise I do get nil.

The work around at the moment is:
              check whether the switch is defined - if not create a dummy L01 - read its index - delete L01

### Expected Behavior

I expect that I always get L01 = 36 (at least on a x9lites) because that's the reserved index for L01.

If it is really necessary to return the result whether the switch is defined  then I would prefer to have two return values - one for the Index and one (true/false) whether the switch is defined or not.
The number of the switch index L01 does always exist and therefore  should always be returned as a value.

I think this behavior does make sense for hardware switches. If for example a radio does not have a switch "SD" then there is also no index number reserved for the switch in the index list. So nil is ok then.

### Steps To Reproduce

Write small Lua  with 

    getSwitchIndex("L01") 

and check for nil (if L01 is not defined in the model)

### Version

2.8.0

### Transmitter

FrSky X9 Lite / Lite S

eshifri, 2023-01-22:
Isn't it better to use getSwitchIndex() for all LS's you need?
Calculating the index from L01 you are relying on certain ETX internal storage structure that is not guaranteed to be the same in the future.

ChrisOhara57, 2023-01-23:
> Isn't it better to use getSwitchIndex() for all LS's you need? Calculating the index from L01 you are relying on certain ETX internal storage structure that is not guaranteed to be the same in the future.

in general you may be right.
But I do use this method inside the block of logical switches only.
So the risk exists only theoretically.

On the other hand it is recommended to use this function only rarely because it is time consuming.

And there is a even more relevant reason to do it this way:
--- I've written a set of scripts about doing settings for Trainer/student flying.
--- They are designed for OpenTx but they do in fact also run on EdgeTX using the same magic number.
--- To ease maintenance I use one script doing all these definitions of 5 "magic numbers". The rest of the scripts load this "library". So implementation of new radios or changing in numbers is quite easy.
--- What I've done now is extending this script with an if clause. So if EdgeTX version is 2.6 or above or OpenTX version above ... then it uses the new method.

ChrisOhara57, 2023-01-23:
added the same request for getSourceIndex("...")

For source "CH1" I can use the same workaround ( creating mixer CH1 -  reading index of CH1  -  deleting Ch1)

But for "TR1" this workaround doesn't work.  You have to turn Trainer mode "ON" in the radios menu to get it work.
Doesn't makes sense when you are doing scripts to be used by other people too.
So the only workaround at the moment is a fallback to the old system of calculating with magic numbers.
In this case it is: index of "TR1"  = index of "CH01" - 16

ChrisOhara57, 2023-01-27:
PS: If you do not want to change it then **please** do clarify  in the LUA Reference Guide for those functions.

eshifri, 2023-01-27:
@pfeerick it is your call.

pfeerick, 2023-01-27:
On anything Lua I usually defer to @jfrickmann and @JimB40 (on holiday so may not respond for a while) as they are the Lua guys. I have this feeling it was intensional as it also supposed to indicate if the source exists, but I could be confusing with a different function.

ChrisOhara57, 2023-01-28:
If the they decide to alter the function I would propose to add a second return value.
So existing scripts would not be effected.

like:    returnValueOne, returnValueTwo  = getSwitchIndex("L01")

so a call:   returnValueOne  = getSwitchIndex("L01") would have the same result as it is now

Why the optional returnValueTwo  would return also a switch number if the switch is possible but not defined at the moment

JimB40, 2023-01-28:
To find your "magic number" of the first logical switch use old getFieldInfo() function or from ETX 2.8 newer function getSourceInfo() that an alias to old one
 
Syntax is getSourceInfo("ls1") and you get table with key 'id' that holds your magic number, wheater LS is defined or not.
https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getfieldinfo-1

ChrisOhara57, 2023-01-28:
I have an x9liteS. (same on every other type of radio)
Using the function getfieldInfo("ls1") or getSourceInfo("ls1") do both return the .id  = 95.
This is the correct ID to feed getValue()  the number of the source of "ls1".

But that's not what I am searching for. I want the index of the logical switch in the **switch list** to feed the parameters of functions like model.setLogicalSwitch(), model.setCustomSwitch(), model.insertMix(), model.insertInput().

The function getLogicalSwitch("L01") does return correct value  which is 36
The only (small drawback) is that the switch "L01" has to be defined, otherwise you get nil.
That is not really a problem but it took me hours to find out why it didn't work on some models and didn't on others.

ChrisOhara57, 2023-01-30:
> To find your "magic number" of the first logical switch use old getFieldInfo() function or from ETX 2.8 newer function getSourceInfo() that an alias to old one
>
> Syntax is getSourceInfo("ls1") and [...]

I have an x9liteS. (same on every other type of radio)
Using the function getfieldInfo("ls1") or getSourceInfo("ls1") do both return the .id = 95.
This is the correct ID to feed getValue() the number of the source of "ls1".

But that's not what I am searching for. I want the index of the logical switch in the switch list to feed the parameters of functions like model.setLogicalSwitch(), model.setCustomSwitch(), model.insertMix(), model.insertInput().

The function getLogicalSwitch("L01") does return the correct value which is 36
The only (small) drawback is that the switch "L01" has to be defined, otherwise you get nil.
That is not really a problem but it took me hours to find out why it did work on some models and didn't on others.