EdgeTX/edgetx #3084: Wireless trainer using DSM protocols

Dave-thirteen, 2023-01-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Companion and Edge 2.8.0 wireless trainer problem

The wireless trainer sets up and works as expected in EDGE  2.7.1

TX16S with external MPM. Both are on 1.3.3.20

The MPM is the Jumper 4 in 1 and does NOT have the rotary switch. Not sure whether 128k or not.
It has worked with OPENTX apart from a small problem with the latest OPENTX  .15 firmware.
Which was,
Model set up for wireless trainer and bound using DSMX for master and DSM_RX for trainer.
This shows up as jittering on some channels (see posts #23905 and mine at #23907 in the DIY Multiprotocol TX Module thread. (never got this resolved as V.14 worked ok).

The problem with EDGE 2.8.0 is that I cannot select the DSM_RX protocol.

If I write to the TX a model created using Companion that has internal MPM set to DSM and external to DSM_RX then
the wireless trainer works as expected but...

In the TX's 'model settings' external MPM page there is only MULTI showing (no DSMRX).
If I click on the pull down to the right of the MULTI box there are less protocols than the internal MPM - about 68 show.
Also the ones showing are not in alphabetical order - for example
Internal      External
DEVO        DSM
DM002      DEVO
DSM
DSM_RX

DSM_RX does not show in the external MPM pull down and the external protocols are not in alphabetical order.
(I was used to DSM_RX being further down the list but it's not there).

If I set either (or both) MPM's to OFF then power down and power up the tx.
When I select a MPM that was set to OFF, I get a message "MPM SCANNING PROTOCOLS"
Once that has completed ALL the protocols seem to be there and I can select DSM_RX on either MPM and protocols are in alphabetical order, plus wireless trainer works.

In companion 2.8 I get a similar problem when running the simulator!

If a model is set up in companion for both radio MPM's to be OFF and the simulator is then run.
Select the model setting page where both MPM's are now showing as OFF.

Going to internal or external MPM, selecting MULTI and then using the pull down there is only DSM and DEVO in the table (also in wrong order) and some protocols are missing.
There is no DSM_RX option.

I have tried setting internal MPM to DSM_RX and external to DSM in a model loaded from Companion and wireless trainer works, but has the same lack of DSM_RX in the pull down. In other words the problem is the same whether the internal MPM is master or slave.

Maybe I'm doing something wrong, so any help would be appreciated.

Thanks 
Dave.

### Expected Behavior

To see DSM_RX protocol in the protocol list!

And be able to select it on the model settings page.

### Steps To Reproduce

As detailed above.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-01-23:
Does it also say "Module Status: No MULTI_TELEMETRY detected" for the external module? As it sounds like your module is not actually talking to the transmitter for some reason, hence ETX is unable to actually get the list of supported protocols from the module, and falling back to an internal static list, which does not have all the protocols listed (and can never be entirely correct, as the protocol/subprotocol support varies from version to version of the MPM firmware). This is also why the simulator will behave the same, as it only the radio firmware being simulated, not the MPM module also. 

btw, the external module should still be doing a protocol scan (that progress bar) on every startup, but you'll only catch the progress bar if you go into model settings -> external RF very quickly on startup. 

So just to be clear - are you saying that if the external MPM is enabled when you turn the transmitter on, when you go into the module settings `DSM_RX` is not showing. i.e. for you

1. With a model that has the External RF set to MULTI, turn the transmitter on
2. Go into Model Setup, External RF
3. Select the second dropdown (for protocol)
4. DSM_RX will not be in the list
5. Change MULTI to OFF, power cycle the radio and repeat 1-2. 
6. Change 'OFF' to 'MULTI', and wait for protocol scan
7. DSM_RX will now be in the list

Dave-thirteen, 2023-01-23:
Hi Peter,
Many thanks for taking the time to come back to  me.

Your sequence 1-7 above is exactly correct.
I set a model in Companion where the external MPM was OFF and loaded it
into the tx.
1) On the TX I clicked  external RF
2) Screen shows OFF as expected.
3) Selected MULTI, got message No MULTI_TELEMETRY detected for a split
second then status shows as V1.3.3.20 TAER
4) Pull down shows FlySky. . There is no DSM_RX in the pull down list.
5) Power cycle TX. , go to External RF and Multi and FlySky still shows.
6) Use pull down and DSM_RX now shows, can be selected etc.

1) I tried setting a model to FRSkyRX for the external RF in Companion and
loaded it in.
2) External RF shows Multi, FrSkyRX and "no multi telemetry detected"
3) Clicked the protocol pull down, got beep, screen brightness went down
and no options showed up.
4) Downloaded a model with both MPM set to off.  In external RF it showed
OFF,
5) turned on to Multi. FlySky shows but when I click the pull down I get a
short beep, screen brightness down and no options shown again.
6) Set a model in Comp with External set to DSM_RX and loaded it.
7) Now get Multi with DSM_RX showing but clicking on the pull down I get
the beep, low brightness and no options. Shows "no multi Telem" message
8) Had to power cycle tx to get it to work!.

* This is also why the simulator will behave the same, as it only the radio
firmware being simulated, not the MPM module also.*

In the sim, when I select either radio module there are limited protocols
and no DSM_RX.
If it's only the radio firmware, why is this?
In the OPENTX Companion sim I can select all of the protocols for either
MPM.

The external module works flawlessly with OPENTX  V14 but has some problems
with the latest V15 as mentioned in my GIThub post.

*btw, the external module should still be doing a protocol scan (that
progress bar) on every startup, but you'll only catch the progress bar if
you go into model settings -> external RF very quickly on startup.*

I don't see it when selecting external RF that was set as OFF, but it does
show when selecting internal RF that was set as off (does it as soon as I
select multi) - the first time.only

WOW, that's a lot and again I thank you for replying.

I can do any tests that will help you out.

Thanks and kind regards

Dave.

Dave-thirteen, 2023-01-28:
Hello Peter,

I wonder if you managed to duplicate the problem I'm having with the
External RF?
Are there any tests / suggestions you can make?
Thanks
Dave.

pelomi, 2023-07-03:
I have the exact same problem. Is there any solution yet?

gagarinlg, 2023-07-03:
Did anyone of you try 2.8.4 or 2.9 RC1?

pelomi, 2023-07-03:
I am on 2.8.4

pfeerick, 2023-07-04:
Have either of you tried updating the firmware on your MPM? v1.3.3.0 is needed for protocol scanning to work. Also, as part of 2.9RC1 which has just been released, the built-in MPM lists have been updated, so match all the protocols supported for the MPM as of [v1.3.3.20](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.3.20) (released in Oct 2022).

pfeerick, 2023-07-04:
>>In the sim, when I select either radio module there are limited protocols
and no DSM_RX.

>If it's only the radio firmware, why is this?

If the MPM is not talking to the transmitter for any reason, and is thus unable to tell the radio what protocols the version of MPM firmware it is running actually supports, there is a fallback list which is shown instead. It can also be browsed if the module is not connected. 

2.8.4 has DSM_RX missing from the internal fallback list, but 2.9 does have it present, so you will be able to pick it from the list now. What does confuse me though is since you got the module status of "V1.3.3.20" (with TAER channel order) ... it should have also done the protocol scan (unless it did it already against the internal module if that was on). Even if you do select it, it sounds like something is wrong with the module (old firmware, pins not lined up, faulty, etc), and other functions may not work.

mha1, 2023-07-04:
I'm not sure if my comment is helpful as I don't really understand the problem here. 

Using the above sequence on my TX16s 2.8.4, internal MPM, external MPM both on 1.3.3.20. 
-     With a model that has the External RF set to MULTI, turn the transmitter on
-     Go into Model Setup, External RF
-     Select the second dropdown (for protocol)
-     DSM_RX will not be in the list
       -> DSM_RX shows
-     Change MULTI to OFF, power cycle the radio and repeat 1-2.
-     Change 'OFF' to 'MULTI', and wait for protocol scan
-     DSM_RX will now be in the list
      -> DSM_RX shows

Additional tests 

1. internal MPM off, external MPM on, but not connected
- External RF Multi settings: no DSM_RX, No  MULTITELEMETRY detected, as expected

2. internal MPM on, external MPM on and connected
- External RF Multi settings: DSM_RX selectable, V1.3.3.20 TAER, as expected

3. internal MPM on, external MPM on, but not connected
- External RF Multi settings: no DSM_RX, No  MULTITELEMETRY detected, as expected

4. internal MPM on, external RF on and connected
- External RF Multi settings: DSM_RX selectable, V1.3.3.20 TAER, as expected

pelomi, 2023-07-04:
I recorded it. I think it has something to do with changing the model.
https://www.dropbox.com/s/btozo1muqft437c/dsm_rx.mp4?dl=0

mha1, 2023-07-04:
Thanks, I was able to reproduce it:

Precondition:
- working external MPM in module bay
- turn radio on
- create two empty models Model1 and Model2
- select Model2
- turn radio off
- turn radio on
- edit Model2 for External RF to be MULTI / DSM_RX
- turn radio off
-> Model 2 with External RF MULTI/DSM_RX is now the active model

Test sequence:
- turn radio on -> Model2 starts up 
- check External RF --> subprotocol selection edit field shows DSM_RX. DSM_RX is in the list
- select Model1
- select Model2
- check External RF --> subprotocol selection edit field is empty, DSM_RX is not in the list
- turn off radio
- turn on radio
- check External RF --> subprotocol selection edit field shows DSM_RX. DSM_RX is in the list

There is no difference if after switching from Model1 to Model2 External RF is set to OFF and back to Multi. DSM_RX is not in the list. Doing the same sequence with the Internal MPM works fine.

It looks like the scan after model change goes wrong or is omitted for the external module (only). This will load the fallback list which doesn't contain DSM_RX in 2.8.4 leading to an empty edit field and DSM_RX not selectable any more until a power cycle. I think DSM_RX is still active and working ok but not shown in the edit field due to the failed or missing scan.

2.9 and 2.10 do show DSM_RX for the same test procedure but probably the updated fallback list just hides the underlying problem.

gagarinlg, 2023-07-04:
When changing models the module is switched off and then on again. Maybe it enters a brown out situation, when ist is not switch off long enough.

3djc, 2023-07-04:
It is unclear if it is on bw or colorlcd. Colorlcd has a process where the full module protocol list is exchanged, maybe there is a glitch with that when changing model

mha1, 2023-07-04:
Well, I can reproduce it on my TX16s, so colorLCD. And it happens on a different external MPM than the problem finder's. Mine is a RadioMaster 4IN1

pfeerick, 2023-07-05:
The initial commenter was on 2.8.0, with TX16S. So with that case, it was most certainly colorlcd specific ;)

mha1, 2023-07-05:
Please check out #3763 for the root cause analysis.

mha1, 2023-07-05:
@pelomi @Dave-thirteen 

Please find 2.8.4+ including the fix for your issue here: test https://github.com/EdgeTX/edgetx/suites/14072040458/artifacts/786613541 
I'd appreciate you testing this.

Dave-thirteen, 2023-07-05:
Thanks Michael. I'm only able to test this tomorrow, but will let you know.
Thanks for all your efforts with Edge.

Dave.#13

Dave-thirteen, 2023-07-06:
Michael,
Just tried your changed firmware and it appears to work OK.
When selecting the external module for a plane that had this turned off I
see it scans the protocols.

I created a new model in Companion with both RF modules not defined and
loaded it into the TX
I see that the internal module defaults to FRsky but the external does a
scan first when turned on to MULTI.
I also note that if the internal module is set to off, then the radio is
turned off and on, then when I go to internal module it does do a scan
after selecting multi.

Looks like you sorted this problem out. Many thanks for taking the time to
have a look at this.

Hope that change makes it into the 2.9 that I see is coming soon.

Thanks to everyone who contributed.

Cheers
Dave#13

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2023-07-10 by pfeerick

mha1, 2024-04-03:
@heyfredb you should update to 2.9

heyfredb, 2024-04-03:
Ah yup! That also fixed the glitchy behavior I was seeing. Smooth as a ball bearing on glass now. Thanks.

PR #3763 (2023-07-10): fix: reset proto scan software timer to MULTI_PROTOLIST_START_TIMEOUT
... if timer already exists (from previous scan)

fixes #3084

changes:
Protocol scans use software timers to fetch the MPM protocol list at 100ms rate but with an initial rate of MULTI_PROTOLIST_START_TIMEOUT (3000ms) for the first poll to allow the MPM to finish its startup sequence after power on. Switching models back and forth power cycles the MPM. As the software timer already exists it is not instantiated again but reused. but with the rate it was left, i.e. 100ms, not the required 3000ms to allow the MPM to settle. R [... cut]