EdgeTX/edgetx #3086: Companion - colorlcd - GVs missing from Inputs source list

pfeerick, 2023-01-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I noticed when looking at one version of a FM setup I use that GVs are not being shown on the GVs source list for colorlcd. It seems this capability is only valid for colorlcd - B&W don't seem to have this, which is possibly why this is missing. This means if I were to read/write the model the setting would most likely be trashed. 

### Expected Behavior

That GVs appear in the Inputs source list (between Channels and Telemetry entires), like on the transmitter/simulator.

### Steps To Reproduce

1. Configure a Input on TX16S with GV as the source
2. Read that model with Companion
3. See that the Inputs source field is blank in Companion

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Present on both release and main build of firmware/Companion.

Attached is the model file I noticed this on. 
[Skysurfer-ELRS.zip](https://github.com/EdgeTX/edgetx/files/10478329/Skysurfer-ELRS.zip)

elecpower, 2023-01-23:
From a quick check GV is not an available Input source for colorlcd in OTX so appears added in ETX.

Questions:
- is there any difference in the mixer engine code between radios types?
- if not, why are GVs not available on B&W edit ui?
- GV is not available as a source in Mixes edit ui for colorlcd, why not if available for Inputs?
- is GV source for a Mix handled in mixer engine code?
- etc

elecpower, 2023-01-25:
Using the X9D+ libsim I can confirm that:
- it does load the yml input GV source
- the ui does not let you edit an input source to a GV but does display
- the mixer engine does not discriminate between B&W and color.

Based on these results there needs to be:
- a fix for B&W radio edit ui
- PR #3097 should be modified to support GV as input source for ALL radios

[image]
[image]
[image]
[image]

closed 2023-03-16 by pfeerick