EdgeTX/edgetx #3088: Curve Edit page / x-axis and y-axis values have different alignment

mha1, 2023-01-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Standard: all x-axis values are center aligned, y-axis values are left aligned. 
Custom: changeable x-axis values are left aligned, non-changeable x-axis values are center aligned, y-axis values are left aligned

[image]
[image]

### Expected Behavior

all x/y values have the same alignment, e.g. left aligned.

### Steps To Reproduce

1. open Model Curve settings
2. add curve
3. observe different alignments
4. change to Custom curve
5. observe different alignments

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

present in current main

philmoz, 2023-01-23:
Left aligned and center aligned examples below - which do you prefer?
(Note: it is not possible to perfectly align everything with center alignment.)

Left:
[image]

Center:
[image]

mha1, 2023-01-23:
Thanks for looking into this. Centered looks good to me.

PR #3094 (2023-01-24): fix(ui) - Center align the text in the number input boxes for the curve editor.
Fixes #3088 

Set the alignment for the curve editor number input boxes to be center aligned.

Make the X & Y numbers align as closely as possible.
closed 2023-01-24 by pfeerick