EdgeTX/edgetx #3089: X-Lite Pro's S1/S2 have no responses in edgetx 2.8.0

kaji331, 2023-01-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

No signal responses in main screen, input/mix/output menu for S1/S2.

### Expected Behavior

In 2.4.0, I can use S1/S2 wheels for my X-Lite Pro.

### Steps To Reproduce

1. Completely replace your SD card's content with 2.8.0 bw 128x64 zip
2. copy xlites-f6d140e.bin to sd card's FIRMWARE directory
3. press cross button and start transmitter, then write the firmware

### Version

2.8.0

### Transmitter

FrSky X-Lite / S / Pro

pfeerick, 2023-01-24:
Have you tried calibrating them since they're an analog input? i.e. RADIO/SYS -> Hardware -> Calibration, centre them when centring the sticks, and then move them full range when moving the sticks.

kaji331, 2023-01-24:
> Have you tried calibrating them since they're an analog input? i.e. RADIO/SYS -> Hardware -> Calibration, centre them when centring the sticks, and then move them full range when moving the sticks.

Thank you very much!

closed 2023-01-24 by kaji331

Talha909, 2023-07-01:
> > Have you tried calibrating them since they're an analog input? i.e. RADIO/SYS -> Hardware -> Calibration, centre them when centring the sticks, and then move them full range when moving the sticks.
> 
> Thank you very much!

I have the same issue. did you find the solution. I already try the calibration screen. But nothing is now in calibration screen. Like there are alwasys two bars between the joystick but they are missing now.

pfeerick, 2023-07-01:
> But nothing is now in calibration screen. Like there are alwasys two bars between the joystick but they are missing now.

With what firmware version?