EdgeTX/edgetx #3098: Curve monitoring broken in inputs. Acts strangely in curves.

Kevltan, 2023-01-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Curves have dropped to the bottom of the display window in inputs, cross hair marker appears to follow the correct path but not on the curve displayed which is to low. Also does not seem to work right in curve display - cross hair follows both gimbal inputs (even though its on the throttle only) and jumps to centre sometimes.

### Expected Behavior

none of the above

### Steps To Reproduce

as above

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.9.0 latest nightly 24-01-22

[image]

pfeerick, 2023-01-24:
> Since this firmware update Curves have dropped to the bottom of the display window in inputs, cross hair marker appears to follow the correct path but not on the curve displayed which is too low.

I can repo this - in the inputs screen both custom and 'normal' curves are being shown offset.

> Also does not seem to work right in curve display - cross hair follows both axis inputs (even though its on the throttle only) and jumps to centre sometimes, this last issue is in previous nightly firmwares too. Is there even supposed to be a cross hair in curves as there isn't in 2.8 ?

No, it doesn't "follow both axis inputs" - if you go specifically into the curve editor, rather than via an input/mix, it will be activated by whatever stick you move, since the curve is not input/channel specific. And yes, it is supposed to be there... this is a recently added feature bringing feature parity with B&W screens which did this already (as well as adding a little bit more) - #2580

closed 2023-01-25 by pfeerick

Kevltan, 2023-01-25:
Excellent 👍

PR #3099 (2023-01-25): fix(ui) - Inputs - curve line is offset
Fixes #3098

Fix size for curve preview in model input edit view.