EdgeTX/edgetx #3101: Flysky EL18 switch C adnormal in Special Functions

Taical, 2023-01-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Special Functions, using switch C at middle and down position not affect normal, for example : Play Sound, Play Track, Play Value
But in Mixes, setting switch C for channel will work ok

### Expected Behavior

Switch C work normal in Special Functions like other switches

### Steps To Reproduce

[image]

### Version

2.8.0

### Transmitter

Flysky EL18

Taical, 2023-01-25:
Just checked in Radio Settings -> Hardware -> Switches, SC is configured as 3 POS
When set SC for a channel in MIXES, it transitions through all three states (-100%/0%/100%) for sure

pfeerick, 2023-01-25:
Perfect... so I think it is just that the SF code isn't detecting it properly. 

@richardclli Does your EL18 have the 3POS populated? I didn't get the extra switches with mine so don't even know how the wiring is configured or if I can just swap SWC for SWA until I open it up again.

richardclli, 2023-02-07:
@pfeerick I think I have updated and verified the switch defaults with Flysky, they tested it as well.  The settings should be correct when one init the radio.yml from scratch.  It may be wrong however, if the radio.yml is in place and comes from another radio.

Here is my switch config for reference:

switchConfig: 
   SA:
      type: 2pos
      name: ""
   SB:
      type: 3pos
      name: ""
   SC:
      type: 3pos
      name: ""
   SD:
      type: 2pos
      name: ""
   SE:
      type: 2pos
      name: ""
   SF:
      type: 3pos
      name: ""
   SG:
      type: 3pos
      name: ""
   SH:
      type: toggle
      name: ""

pfeerick, 2023-02-08:
If you have SC populated as a 3POS, can you see if you can replicate this issue then - it is related to a special function not detecting SCmid and SCdn positions ;)

richardclli, 2023-02-09:
@pfeerick I have once tried similar problem in PL18, I used a toggle switch to reset the timer, but found it is not functioned properly, I am thinking if it is a BUG in EdgeTX.  But I finally found that the toggle switch is not a 1 way toggle (2 pos) but a 2 way toggle (3pos), and after I change the definition to 3 pos, the problem resolved.

richardclli, 2023-05-10:
Duplicated #3560 , same problem.

closed 2023-05-10 by richardclli