EdgeTX/edgetx #3106: "!" entries missing when using PgUp/Dn to preselect switchgroup

ParkerEde, 2023-01-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If I am in a switch selection, e.g. in the special functions, and select a group there with the PgUp/Dn key, I have no possibility to select the entries with the "!"

before using PgUp/Dn:
[image]

after using PgUp/Dn:
[image]

### Expected Behavior

with PgDn/Up preselection, the "!" entries should also be selectable.

### Steps To Reproduce

1. open special function
2. edit an SF
3. Enter switch selection
4. Press PgDn/Up button to preselct a switchgroup
5. try to select !SA- or any else "!" entry

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

all versions

pfeerick, 2023-01-25:
While that is correct, the workaround to then get inverted entries is to either long ENTER or long press on the field once you've selected the "normal" version of it. I'm not sure the best way to resolve this... might leave it for @JimB40 to mull over... as adding the inverted version of all the switches effectivley doubles the lengths of the lists.

eshifri, 2023-01-25:
I would say the long press is a preferred way to handle this: the switch lists are long enough even without inverted versions.

ParkerEde, 2023-01-29:
The ENTER long press is also quite practical and a good solution. The only problem is that no one will figure it out unless they are told. Should we possibly give a hint somewhere?

philmoz, 2023-12-13:
This can probably be closed (PR #4104)

closed 2023-12-13 by pfeerick