EdgeTX/edgetx #3107: From switch L on wards, settings are lost after reboot on X9E Hall radio

Bratski, 2023-01-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Activating the switches from "switch L" on wards, the settings are lost after rebooting the radio. 

### Expected Behavior

Not loosing these settings after reboot.

### Steps To Reproduce

System menu, hardware,
Activate switches from L on wards, chancing settings from None to Toggle, 3Pos or 2Pos.
Turn off radio
Turn on radio
Settings lost, but just saved until "switch K"
  

### Version

2.8.0

### Transmitter

FrSky X9E / X9E Hall

gagarinlg, 2023-02-19:
confirmed on a non hall X9E

Gerold68, 2023-02-19:
Please take also a look into https://discord.com/channels/839849772864503828/1061346267646660768/1061346267646660768

I am quite sure that all these issues have the same root cause.

Regards
Gerold

gagarinlg, 2023-02-19:
Thnks, i will do that. We specifically baought a used X9E (non hall) to have something to test with

PR #3215 (2023-02-20): fix(yaml): the string compare only matched the number of characters f…
…ound in the yaml file, so "SLIDER1" matched "SL" on the X9E

Fixes #3107 

It might make sense to port this also to 2.8, in case we make another patch release for 2.8
closed 2023-02-20 by pfeerick