EdgeTX/edgetx #3108: Trims - custom range and steps (configurable)

Eldenroot, 2023-01-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Trims should have have a configurable range and the Custom Step uses percent as step unit. Now we can just choose from predefined options which are not detailed described.

### Describe the solution you'd like

Check above.

Eldenroot, 2023-01-25:
Configuration independent for every trim.

There should be:
- range
- step (coarse, fine, ... custom - % value)
- independent trim per FM

Kevltan, 2023-01-26:
Think this is an excelent idea and very useful, asked for something similar last week. Hope the developers look into this and adopt.👍👍👍