EdgeTX/edgetx #3109: Timer - enhancements - countdown, beeps etc

Eldenroot, 2023-01-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Another QOL features and additions to this PR and timer function.

https://github.com/EdgeTX/edgetx/pull/2711

- countdown start
- beep for last X seconds
- beep every X second/minute (predefined time)
- etc.

### Describe the solution you'd like

Check above.

factoryroll, 2023-10-12:
I'd also like to suggest some other timer-related features: 

- Count up timer
- Choice of the starting time. I'd like a timer that can count up from 30s to 60s, for example. I know this could probably be achieved with GVARS...but all my GVARS are in use because of the limited number of them. 
- More options for countdown/countup time, not just 5/10/20/30s. Ideally this would be a number entry.

pfeerick, 2023-10-12:
Timers can already count up and down, you just need to set the direction. But the other two points are interesting. Why a timer counting up from 30s to 60s? Why not a 30s timer? 

And, just saying, a lot of this could be achieved with a widget if on ColorLCD.

factoryroll, 2024-05-13:
Sorry for the very late reply, I missed the notification here. 

The reason for the 30 to 60s timer is to match how a human timer in a competition might read the time remaining in your flight. Reading the clock up from 30s to 60s is one example. This allows for better practice. 

Is there a reason to restrict the timer to those 5/10/20/30s options rather than to just allow a user-defined parameter? 

I think ideally all core features are catered by the OS, rather than relying on widgets that might work only on colour screen radios.