EdgeTX/edgetx #3114: Special Function Play Track !1x Ignored with Mix Delay

jmxp69, 2023-01-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With delay up/down in a mix line, a SF audio track is played when model is loaded with !1X as the Repeat Option
Remove delay, and the track is not played when model is loaded.

### Expected Behavior

Seems processing of mix delays comes after evaluation of !1x tracks. 

One thought, could a !Nx option skip playing the first triggered instance?

### Steps To Reproduce

Mixer7 LS Delay Up 0.5s
Mixer7 LS Delay Down 0.5s
L0x = a~x Ch7 30
SFx = L0x Play Track xxxx !1x

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Looks like I picked a bad week to quit sniffing glue.

philmoz, 2024-08-04:
I can't seem to reproduce this in 2.10 - is it still an issue?

jmxp69, 2024-09-11:
No, cannot replicate on 2.10.2.

closed 2024-09-11 by jmxp69