EdgeTX/edgetx #3117: Default to specific model upon starting up the radio

afromanjohan, 2023-01-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I like many others use a radio with an ELRS module in the back. When I am flying I tend to use a model in which I use this ELRS module, when I am done flying I need to remove the antenna for this module. It is not good to power the module without the antenna attached and the module gets power immediatey on startup. When I am at home I have a specific simulator model that uses neither the internal nor the external rf, but when starting the radio after flying I need to connect the antenna and then switch over or rememeber to switch over before shutting it down after flying.

### Describe the solution you'd like

Have an option in probably the models screen that you can select a specific model to be the default model that the radio starts up into.

pfeerick, 2023-01-27:
Ok, so that would probably go in the radio setup screen (since it's not a model-specific setting :wink:) and the question for implementation would then be a matter of if it is just a choice popup menu that lists all the models and you pick one, or somehow shows a version of the model select screen that only lets you pick the model. The first would be the simpler IMO, although it will be a little tedious if you have same-named models (then again, probably serves you right if that is confusing lol).

afromanjohan, 2023-01-27:
Yeah radio setup might also make sense since it would probably be an option to select default or start up into last used (as it is right now). I figured it would make sense to have it as an option in the list you get when you highlight a model where you get the option to make a model a template/make favorite/select the model/etc. Then maybe have a small indicator show the one you have selected. Both makes sense last one would surely be easiest to implement.

pfeerick, 2023-01-27:
Oh... I like that idea :)