EdgeTX/edgetx #3120: Jumper t-pro elrs internal

lucagaff, 2023-01-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating the firmware bro 2.8, I cannot select crossfire protocol in the model page.
For the internal module the possible selections are MULTi or OFF

Is the 2.8 ver compatible with the t-pro with internal elrs module? Many thanks 

### Expected Behavior

Option for crossfire protocol in the internal module options

### Steps To Reproduce

Model page-> internal module

### Version

2.8.0

### Transmitter

Jumper T-Pro

gagarinlg, 2023-01-27:
Go to system-> hardware and select CSRF as internal module

closed 2023-01-27 by gagarinlg

lucagaff, 2023-01-27:
Many thanks , worked !!!