EdgeTX/edgetx #3127: Switch label not showing on jumper t12 monochrome display

frizzle101101, 2023-01-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

switch and pot labels not showing in mix and special functions and vis versa when selecting the switch source, only the switch or pot icon not the switch label text, and there is no quick menu to view the categories logic switches, pots, switches etc. not sure if that menu was supposed to be a thing still like on the x9d opentx, i feel like it has been lost for the t12

### Expected Behavior

I expected to be able to see the SH or such label, not just the switch icon
I also expected there to be a quick menu to view the categories of logic switches or pots and such when you hold down enter on the source selection.,

### Steps To Reproduce

1. on the jumper t12 go to mix
2. open edit on a channel
3. select the source and use a switch or try to hold enter for the quick menu

### Version

2.8.0

### Transmitter

Jumper T12

pfeerick, 2023-01-30:
I think I can reproduce this in the simulator - AFAIK in principle it should behave the same as other 128x64 display radios, and doesn't seem to show that popup menu for mixer source, switch, LS "AND Switch" or SF Switch options, etc.

3djc, 2023-02-11:
I'll try to have a look at it tomorrow

3djc, 2023-02-12:
ok, switch selection seems to work fine, the popup menu doesn't, but thats because it doesn't exist in 9x style navigation that the t12 uses. Unsure if adding it to 9x navigation makes sense

pfeerick, 2023-02-12:
So that means the T12, T-Lite and LR3PRO are the odd ones out there?

3djc, 2023-02-12:
Yes, their navigation is different because of the set of keys they provide

pfeerick, 2023-02-12:
> only the switch or pot icon not the switch label text,

@frizzle101101 That means you need to check if the switch labels have been set to be blank in RADIO/SYS => Hardware - they should appear as a set of three dashes if the default name is being used. If the dashes aren't there, you need to edit the names as there is probably a blank space present.

[image]

pfeerick, 2023-02-12:
True, but it is also a can of worms... as the x9d also only has six buttons but has the popup menu :shrug:

3djc, 2023-02-12:
Yes, but the whole navigation is different. In 9x, you 2d navigate the UI, where with x9d nav, you select rows, and then enter the rows to select collums.

it may be psossible  to develop a similar functionnality for 9x nav, but that would then qualify as an enhancement, and may not be as easy as it looks.

pfeerick, 2023-02-12:
Agreed - that part is certainly not a bug, It would be nice to have the popup menu, as from an end-user perspective, there is no difference in the T12 and say TX12 as far as the UI generally... or the X9D once you take away most of the wide-screen layout changes - i.e. still go into edit screens for inputs and mixes, and then the T12 doesn't have popup, but the others do. Anyway, you know what's a reasonable effort to put into this so I'll leave it's fate in your hands ;)

frizzle101101, 2023-02-12:
> @frizzle101101 That means you need to check if the switch labels have been set to be blank in RADIO/SYS => HARDWARE - they should appear as a set of three dashes if the default name is being used. [...]

that was the issue with the switch labels weird glitch all I had to do was press enter twice, like edit and confirm, and it went the 3 dashes.

SkeeterFPV, 2023-02-12:
I'm having possibly a similar issue with my Radiomaster TX12.  I upgraded from OpenTX to EdgeTX 2.8 and lost my default model from Radiomaster in the process.  I have created a new model and reflashed the radio many times but alas still no switch labels.  I'm completely new to these radios and have only ever flown the a micro rtf kit so perhaps I'm doing something wrong.  I don't have a page under models called "Switches" like the screenshots above.

[image]

pfeerick, 2023-02-12:
Not Models/MDL
> RADIO/SYS => HARDWARE

Try this (I show at the start that the radio actually has three spaces for the name rather than nothing, hence the blank name):

[image]

SkeeterFPV, 2023-02-13:
@pfeerick Thank you so much.  I was going crazy over this.  Glad to know it was a bug and not my doing.

> Not Models/MDL
> 
> > RADIO/SYS => HARDWARE
> 
> Try this (I show at the start that the radio actually has three spaces for the name rather than nothing, hence the blank name):
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/5500713/218342866-9274adae-850b-4854-b3f1-48ff7fc77918.gif) [ ](https://user-images.githubusercontent.com/5500713/218342866-9274adae-850b-4854-b3f1-48ff7fc77918.gif)
> 
> There was a bug related to this that I thought was caught for 2.8.0, or maybe it's in place for 2.8.1.

Daedalus66, 2023-02-14:
My X9 Lite was missing the SA label in a number of menus.

As suggested above, I went to Radio/SYS => Hardware and deleted the dashes for SA and it repopulated. The SA label now appears correctly.

Thank you.

Daedalus66

bd4, 2024-04-27:
This happened to me while reading/writing models into EdgeTX companion with EdgeTX 2.10 RC3. I tried editing radio.yaml manually to fix the extra spaces and editing in companion, and it kept coming back. So seems likely to be a bug with companion?

HippoDan, 2024-05-29:
This problem persists into 2.10 and exists on the TX16s.

gagarinlg, 2025-01-24:
@elecpower did you fix this?

pfeerick, 2025-05-18:
Since this is somewhat documented issue (in release notes, but probably needs to be in a-to-be created FAQ/troubleshooting section of the manual) related to upgrades, I'm going to call this closed unless we can get some exact reproduction steps. I can't reproduce this with a simple read/write of settings with T-Lite, and generally have not noticed this issue when using Companion normally. When it does occur, it seems to either be an upgrade glitch from older version of firmware or deliberate action by the user (spaces in the name in Companion). Both of which can be easily fixed by either removing the spaces in Companion, or on the handset by selecting the field and either clearing it (or on later firmware simply pressing Enter twice on the offending field in Radio Settings/SYS -> Hardware).

closed 2025-05-18 by pfeerick