EdgeTX/edgetx #3131: Problem saving specials functions

kitsen13, 2023-01-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

In companion, specials functions  setting "Set Main Screen" are not saved.

On a line, declare a "Set Main Screen" to anything else than 1 for example '2', save , close model window .
Re-open model  window "Set Main Screen" remains to 1.

### Expected Behavior

Save value.

### Steps To Reproduce

On a line, declare a "Set Main Screen" to anything else than 1 for example '2', save , close model window .
Re-open model  window "Set Main Screen" remains to 1.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

eshifri, 2023-01-30:
How many screens are defined for this model?

elecpower, 2023-01-30:
@kitsen13 @eshifri confirmed as a bug. Wrong field being used for save.

elecpower, 2023-01-30:
Correction not saved at all

PR #3132 (2023-02-03): fix(cpn): Save set main view screen number for special/global function
Fixes #3131
closed 2023-02-03 by pfeerick