EdgeTX/edgetx #3133: EM on TX16S MK1 writing color file on SD

waasnsteffl, 2023-01-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hi 
 
I have the following Emergency Mode Problem on my TX

 TX16S Mk1, 4in1
 2.7.1
 
After some editiing and writing a color scheme back it goes into EM.
Ribbons have been cleaned.
Used 3 different SD cards (8,16,32GB), newly formatted and copied from original resource.
Every cards filesystem has been checked under Windows, every sector read and 
rewritten under Freeebsd.
The first 5 times or so, after creating a new card, you can write a changed
color file to the sd. Then it starts to go into EM. Permanently from now on.
All other changes for model or system writes on the SD working fine.
A binary look in to those written files show no problems in syntax or charackters
compared to the original colorfiles. 
The files who generated a EM are perfectly usable after reboot.
 
Please suggest what i can do. 
As i am a retired informatic system engineer, i am willing to help.
Could changing some hardware help and where to order?
 
Thanks Anton 
 

### Expected Behavior

working write of colorfile

### Steps To Reproduce

see bug description

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-01-31:
There has been some work in this area recently - can you try the latest nightly build and see if the problem is still there.

waasnsteffl, 2023-02-02:
Took me the time and refreshed  bootloader newly with stm32 prog  to 2.8.0. not a nighly at the moment. then mm-stm-serial-taer-v1.3.3.20.bin, added latest LUA, THEMES and MODELS.
same Symtoms, after 3 times editing color and writing to SD, again EM.
Again the color file is working after reboot.

waasnsteffl, 2023-02-03:
installed now 2.9.0 6af2bb7

tx16s-6af2bb7.bin*
mm-stm-serial-taer-v1.3.3.20.bin
 
edgetx.sdcard.version gives 2.9

radio.yml gives
semver: 2.9.0
board: tx16s

 
same problems as before, seems to be even worse
EM when defining widgets, but indeterministic
Widgets seem not to be able to read out TX accu.  
( hope i did not overseen something when defining )
 
theme that i edited, caused EM  included now

again tested r/w every sector on sd, no error
 
btw: when i updated from 2.7.0 to 2.8.0, then loadeded
mm-stm-serial-taer-v1.3.3.20.bin, system hanged in the middle         
of writes.
[theme.txt](https://github.com/EdgeTX/edgetx/files/10578075/theme.txt)

pfeerick, 2023-02-03:
FYI, once you've loaded a MPM firmware, there is no need to write it again... updating or downgrading EdgeTX does not affect the MPM firmware version as that is separate.

When you were editing the theme, did it happen to be the "active" theme rather than an inactive one? As I noticed there is more of a tendency to crash if it were the active theme being saved (and applied, suggesting the issue could be in the code related to refreshing the active theme colours - especially as IIRC the changes had been saved to SD, and the radio booted up fine using the edited colors).

waasnsteffl, 2023-02-03:
complete THE
[THEMES.zip](https://github.com/EdgeTX/edgetx/files/10578153/THEMES.zip)
MES folder as suggested

waasnsteffl, 2023-02-03:
On 2.9.0 I edited both, active and unactive Theme. happens EM appeared after 3 write to SD, and from that on every time...

philmoz, 2023-02-03:
I tried your THEMES folder on my TX16S and I can't reproduce this at all - it is very strange.
I can edit the theme colors as many times as I like and do not get any EM.

Do you have another SD card you could try?

waasnsteffl, 2023-02-03:
I tried 3 diff cards with 2.7.0 with same results.
At the moment  i  have the SD mounted on an up and running 2.9.0.
Under FreeeBSD i have mounted the data partition per usb.
What i test now is the  capability doing a read and a rewrite of the raw sectors of the SD
using dd_rescue, while its in the tx16s.
dd_rescue: (info): Using softbs=128.0kiB, hardbs=4.0kiB
dd_rescue: (warning): Not using sparse writes for non-seekable output
dd_rescue: (warning): infile and outfile are identical!
dd_rescue: (info): ipos:    200704.0k, opos:    200704.0k, xferd:    200704.0k
                   errs:      0, errxfer:         0.0k, succxfer:    200704.0k
             +curr.rate:      411kB/s, avg.rate:      413kB/s, avg.load:-80.+%

may be this shows hardware and or underlying system issues of 2.9.0. but it is extremly slow
in case of no errs, rather points to programming problems.
arent ther log s possible?
what about changing internal hardware as i want to prepare myself for some masterships ?  
do i have a usage/knowlede problem when can i not get the TX accu voltage into a widget?
hope the accu is fit enough for this long write/rewrite

waasnsteffl, 2023-02-03:
have now written/rewritten about 2 Gbs using the tx16s as card reader

[0]root@af[1016][11:55am][~]dd_rescue /dev/da0 /dev/da0
dd_rescue: (info): Using softbs=128.0kiB, hardbs=4.0kiB
dd_rescue: (warning): Not using sparse writes for non-seekable output
dd_rescue: (warning): infile and outfile are identical!
dd_rescue: (info): ipos:   2082816.0k, opos:   2082816.0k, xferd:   2082816.0k
                   errs:      0, errxfer:         0.0k, succxfer:   2082816.0k
             +curr.rate:      410kB/s, avg.rate:      412kB/s, avg.load:-80.+%

no errors......
All Files on SD uncorrupted. taking out the usb plug made the tx16 problemlessly switch 
back to the last task, just in the middle of defining  a new widget.

philmoz, 2023-02-04:
I wonder if this is somehow related to the widgets on the active model?

Can you try setting up a simple model on the radio with no widgets configured and then try editing the themes while the simple model is the active model.

waasnsteffl, 2023-02-04:
Hi,
same with native model, nothing defined, 2nd writing   
of the theme - EM
 
What about the mentioned zero volt in the widget for Bat.
In system voltage is correctable, while zero in the corresponding
widget.
 
used RTos in the 1980ies a lot with Bcpl,
so i do not know how testing is done today,
but isnt there a firmware with logging facility for testing one could use?

philmoz, 2023-02-04:
> What about the mentioned zero volt in the widget for Bat. In system voltage is correctable, while zero in the corresponding widget.

Which widget are you using?

> used RTos in the 1980ies a lot with Bcpl, so i do not know how testing is done today, but isnt there a firmware with logging facility for testing one could use?

You would probably need to hook up a hardware debugger to the radio.

waasnsteffl, 2023-02-04:
was widget/value/batt,    but seems to work after reboot....