EdgeTX/edgetx #3134: sound file will not play with nightly

ror1948, 2023-01-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

sound file timres.wav no longer plays on radio.
it worked before.
i see the bitrate is 128k
it works is i change it to 512k
[timres.zip](https://github.com/EdgeTX/edgetx/files/10551064/timres.zip)

### Expected Behavior

play the track "timer reset"

### Steps To Reproduce

just browse the sdcard in sound dir and select play.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

nighty 2023-01-29

planemann2000, 2023-02-01:
I had the same issue with both of my radios (Eachine tx16s, and Radiomaster tx16s mk2 Max) My work around was to make a new "timer 1 reset wav." that worked for me. So far that is the only sound wav. not working.

ror1948, 2023-02-01:
Yes I did the same.
Just changed the bit rate to 32k instead of 16k

- Ralph

philmoz, 2023-02-01:
The .wav file you uploaded uses the uLaw codec. Support for uLaw and aLaw was dropped a while ago.

Changing the bitrate probably also changed the codec.

ror1948, 2023-02-01:
FYI: the original file was in the release of the amber22 sound pack

pfeerick, 2023-02-02:
It looks like the Amber pack is an odd mix of normal PCM Wave audio files, and uLaw encoded Wave audio. As philmoz pointed out, uLaw and aLaw support was recently dropped as it is very rarely used (and meant we could reclaim some storage space needed for some of radios such as the X9D which have the smallest flash (program) storage. This means that you'll need to re-encode any audio that is improperly formatted, 

[image]

closed 2023-02-02 by pfeerick

pfeerick, 2023-02-05:
Just to elaborate on this further... I just wrote a quick script to parse through all the Amber22 audio files... and would you believe it... _only_ three files are **NOT** standard PCM!? 

Not PCM: ./aclimb.wav
Not PCM: ./timer.wav
Not PCM: ./timres.wav

ror1948, 2023-02-05:
Good job and thanks.
Since the amber22 is a real popular sound pack maybe you should include in your assets for the next RM FW release a copy of amber22+(with full sound pack with repaired 3 files)
I will fix mine anyway and dispute to our club membership.

- Ralph