EdgeTX/edgetx #3140: getvalue numeric seonornames doesn't work

ParkerEde, 2023-02-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I try to read out a sensor with a pure numeric name in LUA widget with getValue, I get no value back. e.g.
Sats = getValue("5100") does not work
If I now change the name to 510a
and then use
Sats = getValue("510a") it is OK.

### Expected Behavior

It should be possible to read data with getValue from numerically named sensors

### Steps To Reproduce

create a sensor with a numeric named sensor.
try to getValue from this sensor

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

actual main

philmoz, 2023-02-02:
From the source code:
_@param source can be an index (number) (which was obtained by `getFieldInfo` or `getSourceIndex`) or a name (string) of the source._

In Lua a string that can be converted to a number, like "5100", is treated as a number - so the code is trying to look up the source by index not name.

getValue(getSourceIndex("5100")) should work (according to the code comments).

ParkerEde, 2023-02-02:
with
Sats = getValue(getSourceIndex("5100"))
I get a
lua:99: bad argument #1 to 'getValue' (string expected, got nil)

philmoz, 2023-02-02:
Digging a bit deeper getSourceIndex looks up names differently than getValue

Try getValue(getSourceIndex(CHAR_TELEMETRY.."5100"))

If you are calling getValue repeatedly in the script (e.g. in the refresh method) then it would be more efficient to save the index returned by getSourceIndex once on startup and then use that in the getValue call.
The name lookup in getValue and getSourceIndex is a linear lookup and the telemetry sensors are at the end of the list.

ParkerEde, 2023-02-03:
Hi @philmoz ,
yes that's it. It works well.
In my case, it is about widgets. Calling getsourceIndex only once at the start poses the problem that if the sensor is not yet ready (e.g. no Satfix) or the model is not switched on and the sensor therefore does not yet supply any data, then it will not be found by getsourceIndex.
So I would have to call getsourceIndex on every run, or use an If condition to ensure that it is only called as long as it is nil. What would you recommend?

philmoz, 2023-02-03:
I haven't tried creating any widgets so there may be better options; but perhaps something like this:
```
	if SatsSensor == nil then
		SatsSensor = getSourceIndex(CHAR_TELEMETRY.."5100")
	end
	if SatsSensor ~= nil then
		Sats = getValue(SatsSensor) -100
	end
	if Sats  ~= nil then
		... use sensor value
	end
```

ParkerEde, 2023-02-03:
many thanks for all your support.

closed 2023-02-03 by ParkerEde