EdgeTX/edgetx #3141: Taranis X-Lite does not power on after installing 2.8.0

SoManyHobbies, 2023-02-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating from 2.7.0 to 2.8.0 the X-Lite briefly shows a storage warning, then will not power on.

Holding the power button to turn on the radio causes the power button to turn green briefly, then the splash screen immediately starts showing the power down animation instead of powering up.

I was unable to find any method to get the radio powered up.

I also tried setting up the SD card for 2.8.0 outside the radio and booting with a good 2.8.0 SD card, but same thing happens.

Reverting back to 2.7.0 resolves the issue.

### Expected Behavior

With the radio off, and USB unplugged.
After holding the power button for a second or so the radio should show the power on splash screen and turn on.

### Steps To Reproduce

Upgrade firmware to 2.8.0 using either https://buddy.edgetx.org/ or using the edgetx-flasher windows app.

After firmware successfully installs the radio will no longer boot.

### Version

2.8.0

### Transmitter

FrSky X-Lite / S / Pro

pfeerick, 2023-02-03:
Can you try flashing the firmware from this github run (it's the latest version of the 2.8 branch, and will form 2.8.1 when released)... scroll to the bottom of the page where it has the downloadable artifact (zip file containing all the various firmware builds).... 
[image] 
... look for the firmware for the x-lite (don't confuse it with the xlites unless you have the X-Lite S) and see how you go with that... there was a firmware overflow issue that was recently picked up which would explain this sort of problem... basically the firmware was too big and would sometimes *sort of work* but not really. 😲

SoManyHobbies, 2023-02-03:
> 

Is there a link to the specific run/branch you wanted tested? I didn't see a timestamp or ID for the run in the screenshot.

pfeerick, 2023-02-04:
Yes, sorry about that, forgot to include the link to the run... but that was indeed the last valid build to try. Thanks for the feedback! :)

closed 2023-02-04 by pfeerick