EdgeTX/edgetx #3145: Unable compile latest firmware in CZ language via Gitpod

japanel, 2023-02-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

When I try compile latest firmware in CZ language via Gitpod, it is end with error.
I am use https://gitpod.io/#https://github.com/edgetx/edgetx/tree/main.
Then I write:
1. cmake -Wno-dev -DTRANSLATIONS=CZ -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=1 -DGVARS=YES -DHELI=NO -DPPM_UNIT=US -DCMAKE_BUILD_TYPE=Release ../
2. make configure
3. make -C arm-none-eabi -j`nproc` firmware

After 90% of compiling stop and failed.

When tried "-DTRANSLATIONS=DE" firmware compilation was success (but I didnt tried flash to radio).

### Expected Behavior

Compile firmware without error.

### Steps To Reproduce

Try compile firmware in CZ language for TX16S via Gitpod.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]

pfeerick, 2023-02-04:
Hm... there's a different code taken for three languages to re-order the words... given the first parameter is a const string that is probably the issue there. 
[image]

closed 2023-02-04 by pfeerick

japanel, 2023-02-05:
Thx, it is ok now.

commit 8d00dc88 (2023-02-04): fix(translation): String casting for CZ/IT/FR
Fixes #3145