EdgeTX/edgetx #3146: Switch SI and SJ power control

MRB-DARKONE7, 2023-02-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No.

### Describe the solution you'd like

It would be great if the software had a link with maybe a logical switch that would allow toggling SI and SJ switches on the TX16S. This would allow to power on and off the LEDs when you want, without wire splicing. It would be awesome if this were possible. 

### Describe alternatives you've considered

1. A DIY switch installed and physically power the lights on and off. 

### Additional context

It would be great if the software had a link with maybe a logical switch that would allow toggling SI and SJ switches on the TX16S. This would allow to power on and off the LEDs when you want, without wire splicing. It would be awesome if this were possible.

eshifri, 2023-02-04:
Can you please explain in more details what are you trying to achieve?
SI and SJ are physical switches and physically connect and disconnect contacts. Obviously logical switch cannot perform this operation.
Are you looking on a LS to manipulate particular port?

MRB-DARKONE7, 2023-02-04:
It sounds to me that what I thought may be possible, is not. I wanted to
use a software switch to toggle the SI and SJ. Based on your response that
is not an option. I wanted thru software turn my gimbal lights connected to
the SI and SJ ports on and off.

eshifri, 2023-02-04:
AFAIK in this mod LEDs are connected to the power line. To manipulate them with SW we need to find some free output pin, add a transistor switch and connect LEDs to it. For SI and SJ these pins are used for input.

pfeerick, 2023-02-04:
Correct. A mod is needed as the gimbal leds simply use those ports to access the always-on power lines. @rotorman did a mod whereby the LEDs are actually controlled by a standalone driver (think it was the https://www.sparkfun.com/products/retired/13911), and it was connected to one of the AUX ports so that a Lua script could control the brightness as well as on/off state.

closed 2023-02-04 by pfeerick

rotorman, 2023-02-05:
Here RCGroups post with more info about the latter: https://www.rcgroups.com/forums/showpost.php?p=48855127&postcount=13970