EdgeTX/edgetx #3154: Build Instructions under Ubuntu 22.04

simos, 2023-02-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The Wiki page with [Build Instructions under Ubuntu 20.04](https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04) covers Ubuntu 20.04. It does so, using [this shell script for Ubuntu 20.04](https://raw.githubusercontent.com/EdgeTX/edgetx/main/tools/setup_buildenv_ubuntu20.04.sh).

I updated the script to work with Ubuntu 22.04. I have switched clang to version 14 (from version 7) because Ubuntu 22.04 does not have a version 7 clang package. 

However, the EdgeTX source code mentions specifically clang 7. Is it required to use clang 7?

[The updated script for Ubuntu 22.04](https://gist.githubusercontent.com/simos/8de45464687d87407041e4c2d2f69500/raw/4530829479584f290a91020a75c4f6e360492704/setup_buildenv_ubuntu22.04.sh).

### Describe the solution you'd like

I would like to update the Wiki to include build instructions for Ubuntu 22.04.

### Additional context

1. Ubuntu 22.04 does not have the 'qt5-default` package. [It is replaced with the following set of packages](https://askubuntu.com/questions/1335184/qt5-default-not-in-ubuntu-21-04): `qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools`.
2. Instead of creating a symbolic link `python` -> `python3`, you can [install the package `python-is-python3`](https://askubuntu.com/questions/1296790/python-is-python3-package-in-ubuntu-20-04-what-is-it-and-what-does-it-actually) which does exactly that.

pfeerick, 2023-02-08:
The build system is only officially supported under 20.04 (i.e. per the docker build environment), so this will be good to have when we do finally move up to 22.04. This will take a while though as it will need extensive testing to ensure that it does not introduce any new issues of its own across the codebase. It's too late in the 2.9 cycle, so I will tag it for consideration as part of 2.10.

elecpower, 2023-02-09:
Qt 6.4 is currently only supported on Ubuntu 20.04 refer https://doc.qt.io/qt-6/supported-platforms.html

lvanasse, 2023-06-04:
I just tried it on my machine and I got those errors message:

[image]

[image]

The two errors above were solved with a simple installation:
`pip install lz4` and `pip install clang`. 

But now I'm stuck at: 
[image]
Although, I do have `libclang1` install on my machine:
[image]

And I've also close and reopen the terminal so that configuration are there. Maybe it's missing a path?

firstqyz8408, 2023-07-29:
> 构建系统仅在 20.04 下正式支持（即每个 docker 构建环境），因此当我们最终升级到 22.04 时，这将很好。这将需要一段时间，因为它需要广泛的测试，以确保它不会在整个代码库中引入任何自己的新问题。在 2.9 周期中为时已晚，所以我会将其标记为 2.10 的一部分以供考虑。

Hi, I built EDGETX in UBUNTU20.4 according to several articles in the WIKI, but I opened the project from QT, it seems that only header files can be loaded, not all files, but there are no errors in the compilation, can you help me?

rotorman, 2023-07-29:
@firstqyz8408 This issue here is for Ubuntu 22.04, not 20.04.
Best, you join EdgeTX Discord and present there some screenshots showing the problems in more detail that you are having.

PR #4410 (2023-12-13): feat: add Ubuntu 22.04 support script
Requires  #4409

Tested building firmware, simu and companion

Will add to wiki once merged

This closes #3154
closed 2023-12-13 by pfeerick