EdgeTX/edgetx #3158: Edgetx touch screen does not work

Ale758122, 2023-02-07:
After updating to version 2.8 my touch screen works at times, as soon as I turn on the tx16s the touch works, but after a touch on the display it goes back to not working, it would be easier to send you a video to show you, how can I do it? With the keys everything works and does not crash, in version 2.7.1 I have not had any problems. Thanks

Mike-i-b, 2023-02-14:
Touch screen stopped working on my tx16s too after 2.8 update. TX is new; only had it a couple of weeks. I don't think it worked at all after update - I've gone back to 2.7 so I can't experiment at present.

gagarinlg, 2023-02-14:
> After updating to version 2.8 my touch screen works at times, as soon as I turn on the tx16s the touch works, but after a touch on the display it goes back to not working, it would be easier to send [...]

You could upload the video to YouTube and share a link

Ale758122, 2023-02-14:
> > Dopo l'aggiornamento alla versione 2.8 il mio touch screen funziona a volte, appena accendo il tx16s il touch funziona, ma dopo un tocco sul display torna a non funzionare, sarebbe più semplice inviarti un video da mostrarti, come posso farlo? Con i tasti tutto funziona e non va in crash, nella versione 2.7.1 non ho avuto problemi. Grazie
> 
> Puoi caricare il video su YouTube e condividere un link

https://youtu.be/UOILRpCxb50

gagarinlg, 2023-02-14:
I think I know what you did.
Take a look at the back light settings and set the mode to keys or both or on.

You have the back light brightness for back light on and off at the same level. When the back light is "off" the Touchscreen is disabled. Since the wakeup is set to only controls, touching screen does not wake up the radio.

Ale758122, 2023-02-14:
No the lighting has nothing to do, the settings are the same, and after using the radio several times it seems to have unlocked itself, in fact the touch works now, but the first time I went out with my drone it had me a strange defect when switching from one drone model to another, namely a momentary block and then release of the voice and a slowdown in model change, now that you told me to check the display lighting, I turned on the radio and the touch works perfectly 🤷🏻‍♂️

gagarinlg, 2023-02-14:
How large are the images you have for you drones? This can cause problems, when they ae not scaled down.

I am pretty sure, that the backlight mode is the problem. At least I could reproduce the exact same behavior by setting the backlight mode to "controls"

Can you please upload an archive with the images, models and radio folders? You can upload files by just dropping them here

Mike-i-b, 2023-02-14:
I just re-flashed 2.8 - touch screen working ok now. Bad flash? User error? Who knows.

Ale758122, 2023-02-14:
> Quanto sono grandi le immagini che avete per voi droni? Ciò può causare problemi, quando non vengono ridimensionati.
> 
> Sono abbastanza sicuro che la modalità di retroilluminazione sia il problema. Almeno potrei riprodurre lo stesso identico comportamento impostando la modalità di retroilluminazione su "controlli"
> 
> Puoi caricare un archivio con le immagini, i modelli e le cartelle radio? Puoi caricare i file semplicemente rilasciandoli qui

I resized the images and the resolution is the same as the other models, same thing on the splash screen images, if they are larger than the original they don't even appear on the display, and not even on the models, I flashed the firmware correctly, both on the Edge tx site via Buddy, and both on the program to install on windows 10 which does the same thing

pfeerick, 2023-02-15:
> Please upload an archive with the images, models and radio folders? You can upload files by just dropping them here

If you do this, we'll be able to look at the settings and hopefully get a better understanding of what is going on. 

Touch can be disabled (by design, not a bug) by both a special function, and also disabled when the backlight mode is set to "Controls" rather than "Keys" or "Both", once the backlight timer has elapsed.

Ale758122, 2023-02-15:
>

Ale758122, 2023-02-15:
> > Si prega di caricare un archivio con le immagini, i modelli e le cartelle radio? Puoi caricare i file semplicemente rilasciandoli qui
> 
> Se lo fai, saremo in grado di esaminare le impostazioni e, si spera, ottenere una migliore comprensione di ciò che sta accadendo.
> 
> Il tocco può essere disabilitato (in base alla progettazione, non è un bug) da entrambe le funzioni speciali, e anche disabilitato quando la modalità di retroilluminazione è impostata su "Controlli" anziché "Tasti" o "Entrambi", una volta trascorso il timer di retroilluminazione.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ale758122, 2023-02-15:
[image]
[image]
[image]
[image]
[image]
[image]

gagarinlg, 2023-02-15:
As I said, backlight us set to "controls"

Ale758122, 2023-02-15:
> Come ho detto, la retroilluminazione ci imposta su "controlli"

it is already on controls

gagarinlg, 2023-02-15:
yes, that is wrong. Set it to "keys", "both" or "ON", but NOT to "controls"

gagarinlg, 2023-02-15:
> > > Si prega di caricare un archivio con le immagini, i modelli e le cartelle radio? Puoi caricare i file semplicemente rilasciandoli qui
> > 
> > 
> > Se lo fai, saremo in grado di esaminare le impostazioni e, si spera, ottenere una migliore comprensione di ciò che sta accadendo.
> > Il tocco può essere disabilitato (in base alla progettazione, non è un bug) da entrambe le funzioni speciali, e anche disabilitato quando la modalità di retroilluminazione è impostata su "Controlli" anziché "Tasti" o "Entrambi", una volta trascorso il timer di retroilluminazione.
> > as you see from the properties they are all resized, the splash is smaller than the templates and is the same size as the original splash
> > [image]

Model images should have those dimensions: 192×114 px

Ale758122, 2023-02-15:
ok i changed to "on" is that ok?
[image]

gagarinlg, 2023-02-15:
> ok i changed to "on" is that ok? [image]

yes, now the touch screen will work correctly, also with 2.8

Ale758122, 2023-02-15:
> > ok ho cambiato in "on" va bene? [image]
> 
> Sì, ora il touch screen funzionerà correttamente, anche con 2.8
ok for the moment thanks, I'll update you later, I go out almost every day with my drone, if any problems arise, I'll let you know immediately, even if I only use the touch when I set the power of the tx or if I change the model of drone, obviously not every day

pfeerick, 2023-02-15:
You might want to read this later so you better understand what those (and other) options do ;)

https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup/additional-radio-settings#backlight

Ale758122, 2023-02-15:
> Potresti voler leggere questo più tardi in modo da capire meglio cosa fanno quelle (e altre) opzioni ;)
> 
> https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup/additional-radio-settings#backlight

thanks for the explanation and thanks for the link, if I hadn't asked you for help, not even the Italian youtuber who makes tutorials and explanations knew anything about it, he wasn't able to help me and what's worse, he abandoned me without giving an answer, not knowing that I can't all things always go well and that some problems can always arise, he didn't even tell me to open a ticket on the Github site, I would have done it immediately, instead I went looking for Edge tx developers through the site! oh well useless people that goes on just to catch likes!

pfeerick, 2023-02-17:
No problem at all. You can also get help via our Discord forum - linked on the front readme page - https://discord.gg/wF9wUKnZ6H

closed 2023-02-17 by pfeerick