EdgeTX/edgetx #3164: IMU support for the TX16S

Kevltan, 2023-02-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would like to be able to use the the tiltX and tiltY on my TX16s.
There is firmware produced in the past by Risto which proved this to work using the LSM6DS33.
The problem is the Adafruit LSM6DS33 which was used has been discontinued.
Adafruit do a current LSM6DSOX which has similar characteristics.
Would it be possible to address and code this IMU to use with the TX16S on Aux1 using I2C communication?
Risto has managed to restore power to the 0v and +5v  pins using forked firmware but I have not been able to get it to communicate with an LSM6DS33 module from my an old X12S mother board. I am not sure if the address/coding are compatible with this.
But saying all that we need new firmware to support a currently available IMU (such as the one above) please 👍😉

### Describe the solution you'd like

as a above

### Describe alternatives you've considered

Adafruit LSM6DSOX

raphaelcoeffic, 2023-02-11:
We would basically need to order one and test with it, should be fairly easy.

raphaelcoeffic, 2023-02-11:
I ordered a LIS3DH as well, so I can check whether or not it would be sufficient, as it is much cheaper than the LSM6DSOX.

Kevltan, 2023-02-11:
The LSM6DS33 works fine in my X12S and now inside my TX16S with the help
of Risto. The LSM6DS33 is off an old X12S board but needed 2 off 4K7
resistors on the PCB module to pull up the SDA and SCL to +3.3v). I use it
to wake the screen up when moving the transmitter at the moment. The IMU is
in the TX16S transmitter mounted on the motherboard PCB.
 Risto uploaded a new version of his firmware yesterday for the IMU
integration into the TX16S but this was somehow messed up in the compiling
process.

Kevltan, 2023-02-12:
> I ordered a LIS3DH as well, so I can check whether or not it would be sufficient, as it is much cheaper than the LSM6DSOX.

Is this just an accelerometer?

pfeerick, 2023-02-12:
Yes. Which I think for the purposes of this is all that is actually needed?

Kevltan, 2023-02-12:
Yes, just for screen wake up then I guess?

raphaelcoeffic, 2023-02-12:
> Yes, just for screen wake up then, I guess? How would you hold a position on the tltX and tltY without the gyro input, wouldn't they just return to  zero when the transmitter held stationary at any angle?

Accelerometer can measure the earth gravitation field very reliably, so no Problem here.

pfeerick, 2023-02-12:
So would a gyro, no? Since it's only measuring the rotational force, rather than linear acceleration. A compass/magnetometer is what returns an absolute position at all times. Looking at the LIS3DH a little more) as I was wondering if it would be senstive enough here, and ST do talk about it being specifically for tilt sensing (both as a simple "portrait or landscape" detection and also for one, two or three axis tilg angle sensing. If you have access to a betaflight or inav flight controller, and look at the values from the accelerometer and gyro, you'll notice it can always tell the orientation via the accelerometer, but the gyro is only active as you apply rotational force.

Kevltan, 2023-02-12:
OK that's very interesting didn't realise that. Always thought that an
accelerometer worked in conjunction with a gyro for example gyro for rate
control and accelerometer for hold control once position established.

pfeerick, 2023-02-12:
Gyro behaves performs the same function in a heli... once it starts rotating, the gyro senses the rotation and how much, and then a counterforce is applied. In the case of a gyro, it's dumb, it's just trying to counter the rotational force and stop that tail spinning. Once you add an accelerometer, now you can self-level, as now you can determine relative position (based on the assumption that when you powered it on, it was on a level surface while it self-calibrated on power on/calibration procedure, etc). They can certainly be fun to play with... one of these days I'll get around to a self-levelling robot using exactly that... works like those hoverboard things that were (are?) all the rage.

Kevltan, 2023-02-12:
Clever things and a great addition to TX16S and should have already have been on the main board as standard like the X12S.

Kevltan, 2023-02-12:
All makes sense now why the IMU unit knows its horizontal relative to the earth. The gyro wouldn't know this only rotational movement.

closed 2024-12-24 by Kevltan