EdgeTX/edgetx #3167: Error loading models in Companion 2.8 Mac

SirDrSpiderpig, 2023-02-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I upgraded to firmware 2.8 on my radiomaster Zorro and then opened the 2.8 companion app.  I tried to read the model from the radio but got the error: "Cannot load Models/model01.yml: yaml-cpp: error at line 269, column 11: illegal map value"
[image]

I'm on macOS Ventura.

### Expected Behavior

I expected to be able to see my models and modify them.

### Steps To Reproduce

1. Turn on Zorro
2. Plug in usb-c cable to data jack
3. Open EdgeTX Companion 2.8 
4. Click "Read Models and Settings from Radio"
5. Error appears

### Version

2.8.0

### Transmitter

Radiomaster Zorro

philmoz, 2023-02-12:
@SirDrSpiderpig Can you upload the problem model01.yml file here please.

ThoFl, 2023-06-26:
I have the same error in the same situation on a radiomaster boxer.
Radio: 2.8.4
Companion: 2.8.4

Cannot load MODELS/model03.yml:
yaml-cpp: error at line 264, column 11: illegal map value

[image]

Here a copy of the model:
[model03.yml.zip](https://github.com/EdgeTX/edgetx/files/11871568/model03.yml.zip)

gagarinlg, 2023-06-26:
Can you please select the model on the radio, change a setting, change it back and try again from companion?
One of the models on the SD card is created with a pre-release Software.

ThoFl, 2023-06-26:
I have in the mean time manually removed the model from the SD card and now it gets past the error.
Thanks for the support!

mha1, 2023-06-26:
Your model03.yml file is corrupt as it has an illegal character in line 263 which is responsible for the radio or companion rejecting this model. I have no answer to the obvious question: why is it corrupt? Did you ever edit this .yml manually? 

[image]

Here's a corrected version of your model file which hopefully will load normally.

[model03.zip](https://github.com/EdgeTX/edgetx/files/11872023/model03.zip)

mha1, 2023-06-27:
You're welcome. Hope it worked,

closed 2023-09-05 by raphaelcoeffic

cengiztanc, 2023-10-16:
I am having this issue, sometimes just one model is affected, other times I have to manually erase almost all models off SD card before Edgetx Companion will successfully read models off radio. Thankfully I have a build backed up on laptop, which I then open in Companion and write to radio. After I fly I come back hook up and bam, models corrupted with little errors.