EdgeTX/edgetx #3186: wrap around in inputs, mixes and outputs: no +button visible

ParkerEde, 2023-02-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the input, mixer and outputs, the + button is not displayed with the first wrap around with the scroll wheel, because the last data set is displayed instead. Only when you scroll in the other direction do you see the +button. Inputs, mixes and outputs are affected

### Expected Behavior

If I jump to the end with the scroll wheel by wrap around, the +button should be visible immediately.

### Steps To Reproduce

1. Open MDL, Tab INPUTS
2. scroll with encoder in Up direction to wrap around to end of table.
3. see, there is the last Input data set, but no + Button visible.
4. Scroll in down direction with the scrollwheel, now you see the +button

### Version

2.8.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

actual nightly

philmoz, 2023-02-13:
On the inputs and mixes pages the '+' button is the selected button when you first load the page, even though it is off screen. When you scroll up you are scrolling to the last row. This is confusing and should probably be changed to select the first row instead.

There is no '+' button on the outputs page that I can see. The 'Add all trims to subtrims' button is selected on first load.

ParkerEde, 2023-02-13:
yes, oututs of course have no + key. Unfortunately I forgot to describe what the problem is here. After the first wrap around, the last line is only half displayed. Sorry

ParkerEde, 2023-02-14:
this PR works fine. Thank you.

PR #3189 (2023-02-22): fix(color) - Fix scroll wheel behaviour on inputs, mixes and outputs pages.
Fixes #3186

Summary of changes:
- Move button creation on inputs and mixes pages so the '+' button is not the first selected button.
- Force line height on outputs page so initial scroll height works.
closed 2023-02-22 by pfeerick