EdgeTX/edgetx #3187: TX16s FRSKY X8M bind mode not retained

gysmitty, 2023-02-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After a successful bind to a Frsky X8M reciever in FrSky X2 mode and either power cycle the transmitter or switch to another model the transmitter reverts back to FrSky protocol and is no longer bound to this receiver.

If I open the transmitter in Companion make the change and send to the transmitter, the setting is retained properly.

The reciever has 2.0.0 firmware(X8R_ACCST_2.1.0_FCC) and the TX16S is running (edgetx-tx16s vers v2.8.0 "Flying Dutchman"

### Expected Behavior

The bind mode will be retained in the TX upon power cycle or model switch and the receiver would remain properly bound.

### Steps To Reproduce

1. Bind to X8M receiver in  FrSky X2 mode
2. Cycle power on transmitter and receiver
T3. The receiver will no longer be bound and the roiginal bind mode reverts to "FrSky"

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-02-14:
@gysmitty What is the receiver ID set to in the module setup?

I recall initially having this issue if the receiver ID was set to 0.

gysmitty, 2023-02-14:
Hey, thanks so much for getting back to me.

I think that it was set to 0. I'll check tomorrow and update the issue if
so.

Geoff Smith

raphaelcoeffic, 2023-02-14:
@philmoz could this simply be a missing “storage dirty” flag?

aivisbush, 2023-04-05:
Hi! I have similar issue.
Device: tx16s
FW: EdgeTX "Flying Dutchman" v2.8.2 (clean install, without any backup cfg)
Scenario:
1. Select Internal module: MULTI - FrSkyX
2. Restart tx16s
Expected: After restart mode stays FrSkyX
Result: Mode is reset to to ESky. Also if you bind e.g XM+, bind is lost

Devs, is there any workaround? Or maybe I should flash back to some older version, but which?

aivisbush, 2023-04-05:
Found solution. Problem was that MULTI module FW was out of date. Updated to the latest, following steps https://downloads.multi-module.org.
And now I cannot reporduce this issue.
I think, this issue can be closed if @gysmitty won't report back...

pfeerick, 2023-04-07:
That would explain why I've never been able to reproduce this, so yes, unless there is further information to suggest this is a bug on our side, this issue will get closed soon. Thanks for the followup :)

gysmitty, 2023-04-09:
Sorry for my lack of response, emails were going to spam folder. I will try the Multimodule firmware update and report back. Thanks to you all who put the time into making this platform better.

closed 2023-09-05 by raphaelcoeffic