EdgeTX/edgetx #3188: FrSky ARCHER PLUS R10+ and SR10+ receiver support

kochpet, 2023-02-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The Frsky R10+ receivers have 10 configurable channel ports, each channel port can be assigned as PWM, SBUS, FBUS, or S.Port. 
At the moment you can only change the channel mapping for the 10 ports.

### Describe the solution you'd like

The possibility to configure all channel ports also to PWM, SBUS, FBUS, or S.Port.

kochpet, 2023-02-14:
Hi Raphael,
thanks for your reply. At the moment you can only configure the ports of the Archer Plus series with ETHOS to PWM, SBUS, FBUS, or S.Port, for this I flashed one of my transmitters to ETHOS.

The reconfiguration of the ports as well as changing the chanel mappings happens with a bound RX under options. I would like to support you in the implementation if you tell me how I can help you!
Attached two screenshots from Ethos.
[image]
[image]

kochpet, 2023-02-14:
@raphaelcoeffic Ok, just to clarify my background. I'm an IT consultant and I work for a company that develops and builds electronic devices. It should be possible that I can provide you with this data.
Can you tell me where exactly I have to attach the logical analyzer. I would use a Frsky X10S ETHOS with an internal ISRM module for this.

raphaelcoeffic, 2023-02-14:
@rotorman can you please help this gentlemen with the pinout of the internal module connector? It is the same as on the TX16S.

raphaelcoeffic, 2023-02-14:
This is the code for the current receiver settings as we know it:
https://github.com/EdgeTX/edgetx/blob/5f171560360ab3982724c2ecc6fba3fc6fe30e16/radio/src/pulses/pxx2.cpp#L397-L419

kochpet, 2023-02-14:
@rotorman on the X10S IXJT motherboard is a 13pin Molex and not a 12pin like at your photo.
Do you have a pinout for that too?
[image]

kochpet, 2023-02-14:
@rotorman
Don't worry already found it
[image]

kochpet, 2023-02-16:
@raphaelcoeffic 
Ok, I'll get the Logical Analyzer on Monday, then we'll see. In the meantime, a possibly stupid question. As I can see at the code you know how to switch the separate port for SBus, Fport......etc on existing receivers. Since the R10Plus registers and binds perfectly and the port mapping also works perfectly, the receiver can also be switched to 7ms PWM mode and telemetry can be disabled, it would not be possible for the reconfiguration of the ports to work with the existing code?
[image]
[image]

raphaelcoeffic, 2023-02-16:
It‘s hard to guess, I don’t have the hardware, so I cannot test, and it feels weird to make blind changes. Let‘s see what traces show first.

kochpet, 2023-02-17:
@raphaelcoeffic 
Can you please check the Saleae logic file, I think this is the sequence we are looking for.

[image]
[Pin1toFBUS.zip](https://github.com/EdgeTX/edgetx/files/10764744/Pin1toFBUS.zip)

kochpet, 2023-02-22:
@raphaelcoeffic Sorry, I zipped the wrong file, here are all the .sal of the various switching processes.
[Analyzer.zip](https://github.com/EdgeTX/edgetx/files/10801411/Analyzer.zip)

raphaelcoeffic, 2023-02-22:
@kochpet did you actually have "Telemetry disabled" activated? It seems bit 7 is always set in your traces, which seems odd.

kochpet, 2023-02-22:
@raphaelcoeffic Yes telemetry was disabled

kochpet, 2023-02-22:
@raphaelcoeffic 
10 Configurable Channel Ports
CP1: PWM / SBUS Out / FBUS / S.Port / SBUS In
CP2-10: PWM / SBUS Out / FBUS / S.Port

raphaelcoeffic, 2023-02-22:
@kochpet it seems we're missing the capability frames in the trace. You should be able to see at some point (possibly when opening the receiver options?) a frame coming from the module starting with `0x7E [size] 0x01 0x06`.

kochpet, 2023-02-22:
@raphaelcoeffic let me check, I make a new trace

raphaelcoeffic, 2023-02-22:
The receiver send `0x70` as its capabilities, so basically bits 4, 5 and 6 are set. Bit 4 is supposed to be "Racing mode" (do you even see the option in the menu?), but we don't know what bits 5 and 6 mean.

```c++
enum ReceiverCapabilities {
  RECEIVER_CAPABILITY_FPORT,                          // bit 0
  RECEIVER_CAPABILITY_TELEMETRY_25MW,
  RECEIVER_CAPABILITY_ENABLE_PWM_CH5_CH6,
  RECEIVER_CAPABILITY_FPORT2,
  RECEIVER_CAPABILITY_RACING_MODE,                    // bit 4
  // missing bit 5
  // missing bit 6
  RECEIVER_CAPABILITY_COUNT
};
```

kochpet, 2023-02-22:
This is the caprure for the High PWM Speed ​​mode
[High PWM Speed.zip](https://github.com/EdgeTX/edgetx/files/10803290/High.PWM.Speed.zip)

kochpet, 2023-02-22:
@raphaelcoeffic Would it be possible that bit 5 and 6 is responsible for switching from SBUS-16 to SBUS-24, you have the trace for it SBUS16-24.sal

raphaelcoeffic, 2023-02-22:
Capabilities and settings are 2 different things! It is not the same bitfield. But yes, the Receiver settings frame, bit 5 is definitely used to switch SBUS to 24CH channel mode. However, only one bit is required, and there is no relation between settings and capabilities, they don't use the same bits and the frames are different.

raphaelcoeffic, 2023-02-22:
Fast PWM was already know and AFAIK already supported. These are the known settings bits (general, not per channel):
```c++
#define PXX2_RX_SETTINGS_FLAG1_TELEMETRY_DISABLED  (1 << 7)
#define PXX2_RX_SETTINGS_FLAG1_READONLY            (1 << 6)
#define PXX2 RX_SETTINGS_FLAG1_SBUS_24CH           (1 << 5) // new
#define PXX2_RX_SETTINGS_FLAG1_FASTPWM             (1 << 4)
#define PXX2_RX_SETTINGS_FLAG1_FPORT               (1 << 3)
#define PXX2_RX_SETTINGS_FLAG1_TELEMETRY_25MW      (1 << 2)
#define PXX2_RX_SETTINGS_FLAG1_ENABLE_PWM_CH5_CH6  (1 << 1)
#define PXX2_RX_SETTINGS_FLAG1_FPORT2              (1 << 0)
```

For the channel output, special values in the output mapping array are used. So far I noted from your traces:
```c++
        // (1 << 5) SBUS in  (0x20)
        // (1 << 6) S.PORT   (0x40)
        // (1 << 7) SBUS out (0x80)
```

Please note that values up to `23` are reserved for normal channel mapping.

kochpet, 2023-02-22:
Ok, how can we continue, try and error?

raphaelcoeffic, 2023-02-22:
Looking at the tandem receivers, it seems SBUS 24CH is a newer feature, so I would assume this is `missing bit 6`, while the "Configurable Channel Ports" (PWM / SBUS Out / FBUS / S.Port) would be `missing bit 5`.

kochpet, 2023-02-22:
Unfortunately, I can't test this with a tandem receiver because I only have an X10S Express transmitter. But we wouldn't need tandem receivers for EdgeTX anyway!

kochpet, 2023-02-22:
@raphaelcoeffic Do you think that you can compile a test version for a TX16S with the current level of knowledge?

raphaelcoeffic, 2023-02-22:
Should be ok. I’ll have something in the next days.

raphaelcoeffic, 2023-02-23:
@kochpet I'm still missing an answer to this question: can you have multiple SPORT pins? (same for SBUS in/out).
Meaning: once you have configured for example "S.PORT" output for pin 1, can you select "S.PORT" output also for another pin? (same question for SBUS/FBUS). This is quite important for the UI.

kochpet, 2023-02-23:
@raphaelcoeffic Yes, you can configure multiple ports, see screenshot. But only Pin1 can also be configured as SBUS In.
Pin1: PWM / SBUS Out / FBUS / S.Port / SBUS In
Pin2-10: PWM / SBUS Out / FBUS / S.Port
[image]

kochpet, 2023-03-07:
@raphaelcoeffic 
Hi Raphael, can you compile a test version for my TX16S. Would like to test whether all the functions really work now.

pfeerick, 2023-03-08:
To access the automatic builds for a PR, first open up the PR page. Under the PR title, you should see a series of tabs (Comments, Commits, Checks, Files Changed). Click on the Checks tab. Then if you are after the radio firmware, click on the link on the left side mentioning firmware. If you want a Companion/Simulator build, pick the one that suits your operating system. Then, under the 'Artifacts' heading will be a link to the zip file containing either the firmware or companion package.

i.e. the PR is mentioned above, as well as at the top of the page, as being #3249, and the latest build run for that is https://github.com/EdgeTX/edgetx/actions/runs/4251600010

However, I will probably need to trigger a new run as that build is about to expire.

kochpet, 2023-03-08:
Hi, I have now tested the new functions for the Archer R10+ with the firmware tx16s-e50d024 and unfortunately have to report that switching the ports does not work at all. Also, the current port status is not displayed in the options screen. I tested with an Xact servo and a FLVS, as soon as the port is reconfigured it's like dead, it doesn't matter whether the port is configured as SPort, Sbus or Fport. If the port is reconfigured back to PWM, it will work normally again.
[image]

kochpet, 2023-03-13:
@pfeerick Will further actions follow in this regard and do you need any information from me.

pfeerick, 2023-03-13:
Raphael has been tied up with real life for the last two weeks so will probably follow up on this soon to chase up on what's needed if this is still something we can debug blind.

raphaelcoeffic, 2023-03-13:
Given the difficulty of doing “remote debugging”, without any possibilities to do any local tests, I think I should order one of these to test myself… that would be another 60-70€ of our donations going to a vendor that does not support us…

kochpet, 2023-03-13:
@raphaelcoeffic I have 2 pieces of the R10+ and at the moment the receivers are almost worthless for me as I certainly won't switch to Ethos. If you send me your address I can support you with one of the two R10+. The experience you make with it is certainly also an advantage for the new RB25 etc., as they have a similar principle of port switches.

jogybaer0815, 2023-03-14:
It would be a great achievement if the Archer R10 with its bus system, especially the F.Bus, were supported by EdgeTX.
If an Archer R10 is needed to achieve this goal, I would also be happy to donate a receiver.
Greetings

DaTwo, 2023-04-05:
Is this still being worked on? The R10+ RX sounds perfect for us gasser pilots.

r41065, 2023-04-18:
FrSky introduces a complete Series of Archer Plus Receiver
( RS Mini, RS+, R6+, GR6+, R8+, GR8+, SR8+, SR10+, R10+ )

hopefully there will be a solution

pfeerick, 2023-04-19:
Real life has intervened, so Raphael hasn't had time to work on this (or much of ETX the last few weeks) ... normal service should resume soon 🤭

gagarinlg, 2023-04-26:
1. FrSky has no interest in EdgeTX, so support for new products might or might not happen
2. EdgeTX is a hobby for us. Some things are way more important than hobbies. Please be patient.

kochpet, 2023-07-27:
@raphaelcoeffic 
Is there any news on the subject or is the work on it completely stopped ?

raphaelcoeffic, 2023-07-27:
> Is there any news on the subject or is the work on it completely stopped ?

I need to get back to it. Probably when I get back from vacations (beg. August).

AlphaCharlyX, 2023-09-07:
Maybe some posts could be a hint here too:

https://github.com/opentx/opentx/issues/9015#issuecomment-1709468832

raphaelcoeffic, 2023-09-07:
> Maybe some posts could be a hint here too

The receiver name table is really not the issue.... What did not work here was the output mode. It's harder to reverse engineer than just finding a string table in a binary (have fun with Ghidra ;-)).

Top of the table:
[image]

Bottom of that same table:
[image]

But you know, neither me nor any other dev from this project does use this gear. So, yes, we could maybe provide some little fan service, you can be sure that as long as nobody actively tests it regularly, it will be very unreliable and incomplete.

Other vendors provide us with the full package:
- sample hardware,
- specifications,
- donations,
- etc...

raphaelcoeffic, 2023-09-07:
@AlphaCharlyX it would help much more if you could find a dev using FrSky/ACCESS willing to support their gear ;-) (*warning: some reverse engineering involved....*)

AlphaCharlyX, 2023-09-08:
Did someone asked FrSky for the needed information? 

I think FrSky 8s interested to sell the Archer plus Receivers to OpenTX and EdgeTX -users as well?

gagarinlg, 2023-09-08:
> Did someone asked FrSky for the needed information?
> 
> I think FrSky 8s interested to sell the Archer plus Receivers to OpenTX and EdgeTX -users as well?

FrSky has abandonded OpenTX, sees EdgeTX as a threat and tries to force people into their closed eco system. A while ago someone from FrSky said, that OpenTX and EdgeTX are old and have a bad software quality. And for OpenTX this is not far from the truth, some parts are not well maintained. But on EdgeTX we are working on those parts, to have a good base for future developments, so that we are able to use STM32H7 MCUs and maybe MCUs from different manufacturers.

FrSky wants everyone to move to Ethos.

If you are able to get infos from FrSky, feel free to share them and maybe someone will implement it. I am pretty sure that they are not willing to share any information. As for me personally, except for then BnFs with Spektrum, I have switched almost everything to ExpressLRS and I do not plan to invest any money or time on FrSky gear, with the exception of testing already supported radios.

AlphaCharlyX, 2023-09-22:
Problem has been solved in OpenTX

pfeerick, 2023-09-22:
@AlphaCharlyX Do you have one of the newer supported receivers? You should be able to try out https://github.com/EdgeTX/edgetx/pull/4077 then ;)

raphaelcoeffic, 2023-09-22:
> Problem has been solved in OpenTX

Including the additional options for the configurable ports? I haven't seen anything related to that, but I might be wrong.

pfeerick, 2023-09-22:
Please keep in mind: OTX has only added *some* support - around being able to OTA flash the Rx IIRC, those changes don't include ability to configure the Rx

pfeerick, 2023-09-22:
Does anyone know if the GR6+ also supports remappable pins? I'm contemplating getting one, else it will be the R10+... looking at the manual for the GR6+ it seems like it has that capability also... 
[image]

kochpet, 2023-09-22:
As far as I know, all Plus series receivers have remappable pins!

3djc, 2023-09-22:
Everything should be available, if something isn't, let me know

3djc, 2023-09-22:
> Please keep in mind: OTX has only added _some_ support - around being able to OTA flash the Rx IIRC, those changes don't include the ability to configure the Rx (which is what this issue is in regard to). So #3249 is still in a state of "need some help testing" before this issue is solved.

Sorry to disagree with that :)

AlphaCharlyX, 2023-09-22:
https://github.com/opentx/opentx/issues/9015#issuecomment-1731043143

I get it from there...

ToniCupra, 2023-09-22:
> Everything should be available, if something isn't, let me know. The issue for pin map was the wrong model id check

I must wait for the nightly version? Sorry for asking (newbie)  :-)

pfeerick, 2023-09-23:
> Sorry to disagree with that :)

We'll fine out... I'm getting the hardware to check :-P

pfeerick, 2023-09-23:
> > Everything should be available, if something isn't, let me know. The issue for pin map was the wrong model id check
> 
> I must wait for the nightly version? Sorry for asking (newbie) :-)

For OTX changes yes. For ETX, niether https://github.com/EdgeTX/edgetx/pull/3249 nor https://github.com/EdgeTX/edgetx/pull/4077 are merged yet so you would need to either using the automated builds avavailable for one or the other depending on what specifically you are wanting to test or build your own frankenbuild [via Gitpod](https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod)

raphaelcoeffic, 2023-09-26:
Gentlemen, the **test** party is going right here: https://github.com/EdgeTX/edgetx/pull/4077

closed 2023-10-09 by pfeerick