EdgeTX/edgetx #3192: Boxer "Play Sound" not working in Special Functions

SBFPV, 2023-02-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

"Play Sound" doesn't work in Special Functions. "Play Track" and "Play Value" work fine. 

### Expected Behavior

I should here the "Play Sound" play when the programmed action is active 

### Steps To Reproduce

1. Program a "Play Sound" to play on the Special Functions page
2. Select the action that should play the selected sound

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster Boxer

### Anything else?

02/14/2023 5f171560

raphaelcoeffic, 2023-02-15:
What do you mean with "Select the action that should play the selected sound"? Could you please upload your model & radio files? (`/RADIO/*` and `/MODELS/*`)

pfeerick, 2023-02-15:
I've just tried the following  on mine, and can't repo this, so will need to see the model files as requested before we can get any further with this. 

As an example, I did this - where SAdn will keep repeating every 2 seconds, and SFdn will beep on press. 
[image]

SBFPV, 2023-02-15:
Thanks pfeerick, that's the exact issue I'm having. They will play fine on my other radios, just not the Boxer. 

Hopefully this contains the info you need [B1.zip](https://github.com/EdgeTX/edgetx/files/10743926/B1.zip)

https://youtu.be/JLBUdEK1XP8

pfeerick, 2023-02-16:
Check your Sound Mode setting (SYS -> Radio Setup -> Sound -> Mode) - it's set to Alarms and you probably want that NoKey ;) unless you only want to hear alarm sounds ;)

SBFPV, 2023-02-16:
> Check your Sound Mode setting (SYS -> Radio Setup -> Sound -> Mode) - it's set to Alarms and you probably want that NoKey ;) unless you only want to hear alarm sounds ;)

Hey it works! Thanks!

closed 2023-02-16 by SBFPV

pfeerick, 2023-02-16:
Fantastic :) Now... one other minor question... what's the origin of the `radio.yml` for that radio... solely edited on that radio? Tweaked with Companion? Just wondering what made all its switch and stick names go triple blank space ... this bug is getting annoying. I'm "this close" to proposing we auto-reset to empty if the field is three spaces...

SBFPV, 2023-02-16:
> Fantastic :) Now... one other minor question... what's the origin of the `radio.yml` for that radio... solely edited on that radio? Tweaked with Companion? Just wondering what made all its switch [...]

As soon as I powered the radio for the first time  I noticed when paging through the screens on the default model that a page had only arrows and no letters for the corresponding switches. This caught me by surprise since all of my other transmitters didn’t need anything done to get the switches to display. After setting everything to - - - it was fine. It’s fixed in the attached photo, but this is the screen I noticed the lack of switch letters 
[image]

pfeerick, 2023-02-16:
Ah, ok, so _possibly_ an issue in the stock settings being shipped in that instance... good to know :) Thanks!