EdgeTX/edgetx #3195: Crash of Simulator when running a lua script with drawRectangle

ggoraa, 2023-02-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

The whole thing crashes when running this code:

```lua

local video_data = {
-- truncated
}

local function delay(time)
    local currentTime = getTime()
    while (currentTime + time > getTime()) do end
end

local function init()
    lcd.clear()
    playFile("/SOUNDS/badapple.wav")
end

local function run(event, touchState)
    lcd.clear()
    for i, frames in ipairs(video_data) do
        local time = getTime()
        for _, coords in ipairs(frames) do
            print(string.format("Rectangle - x1: %d, x2: %d, y1: %d, y2: %d", coords[2], coords[1], coords[4], coords[3]))
            lcd.drawRectangle(coords[2], coords[1], coords[4], coords[3])
        end
        print("frame: ", i)
        -- lcd.drawText(0, 0, string.format("%d", i))
        lcd.refresh()
        while (time + 10 > getTime()) do end
    end
    return 0
end

return { run=run, init=init }
```

...with this error:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               companion [58769]
Path:                  /Applications/EdgeTX Companion 2.8.app/Contents/MacOS/companion
Identifier:            org.edgetx.companion
Version:                ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-15 10:21:16.7341 +0100
OS Version:            macOS 13.2 (22D49)
Report Version:        12
Anonymous UUID:        1F729388-7A6A-BF7A-50B9-394BCB76FBDE

Sleep/Wake UUID:       D1E1C032-012C-4743-BD0D-BD8910980D04

Time Awake Since Boot: 270000 seconds
[... 608 lines cut ...]
USB Device: Flash Card Reader/Writer
Thunderbolt Bus: 
```

### Expected Behavior

Well, not crash and at least show a meaningful error message 🤷

### Steps To Reproduce

1. Go to the simulator
2. Open the script
3. See the simulator shit the bed while trying to run it

### Version

2.8.0

### Transmitter

Other (Please specify below)

### Anything else?

For debugging purposes, here is a file with some video data: [badapple.lua.zip](https://github.com/EdgeTX/edgetx/files/10741500/badapple.lua.zip)

ggoraa, 2023-02-15:
I have not tested this on a real device because the radio doesn't have enough memory even for 20 frames

ggoraa, 2023-02-15:
probable cause is that drawRectangle does not exist, but before this it was just throwing an error about a missing symbol instead of whole on crashing

raphaelcoeffic, 2023-02-16:
I'm using the simpler `simu` to test (does not use Qt) and it "works". By that I mean that it does not crash or causes any bad memory accesses (tested with `-DASAN=YES`; LLVM's address sanitiser).

frankiearzu, 2023-02-22:
That method of delay seems weird "while (time + 10 > getTime()) do end",   the run() function should do work and exit as soon as it can.
Have you try do it incrementally.. i.e,  on every call to run() do only 1 frame?? and if it is too soon to display the frame, exit run() without doing anything?   You will need to keep track of the what is your current displayed frame as well as the last displayed time at script level variables.

ggoraa, 2023-02-22:
That's a good suggestion, thanks! But that's an issue with the _script logic_, and it should definitely not result in the simulator dying
btw now it works fine after upgrading to 2.8.1, so yea that's that

gagarinlg, 2023-02-22:
No Lua script should be able to crash the radio.
But there are a lot of checks missing.

frankiearzu, 2023-02-22:
The challenge with embeded programming is that is not like traditional PC software.  In many embeded systems, they use what is called "Cooperative Multitasking" that is that a running task has to give up the CPU to switch to other tasks.   That was the original intention on OpenTX/EdgeTx.   EdgeTX changed it multi-tasking library to a more modern one, but the LUA programming philosofy shoud still be the same. Short running method calls and not hog the CPU

pierrotm777, 2024-03-31:
Hello frankiearzu,

I look for a method to keep a message on my screen 3s.
I have tried this method without success
```
  local idSA = getFieldInfo("sh").id
  local ss = getValue (idSA) -- ON=1024, OFF=-1024
  if ss > 0 then -- chech SwReset switch
    local timeMs = getTime()
    lcd.drawText(20 , 50, "RESET SWITCH TOGGLED", DBLSIZE + RED + SHADOWED)
    while (timeMs + 3 > getTime()) do end
  end
```

Thanks for your help,

Pierre

frankiearzu, 2024-03-31:
You cannot have delays or long delay loops in LUA.  Every time a refresh it called, it needs to return as soon as posible.You can do a state machine.. with 2 states: Normal and DiplayMsg.You start in Display state,Every time refresh is called, if you need to show a message, you record your start time and switch to display state.In the display state, you display your message, and check if the time is past 3s.. If so, you change your state back to normal.

pierrotm777, 2024-03-31:
Do you have an example how to change the display state ?

frankiearzu, 2024-03-31:
Is just a variable, you assign a value to it. You can say value 1 is normal, value 2 is displaying message.So in your main loop you will have an IF for each of the state (1 and 2) and put the corresponding code for each.

pierrotm777, 2024-03-31:
Ok, thanks