EdgeTX/edgetx #3196: EL18 bind with INr6-HS,Telemetry data problem

lolfling, 2023-02-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[image]

### Expected Behavior

Please check the picture content。

### Steps To Reproduce

1、Bind With INr6-HS
2、View the Telemetry page

### Version

2.8.0

### Transmitter

Flysky EL18

### Anything else?

No response

raphaelcoeffic, 2023-02-15:
Apparently, the sensor data format changed since AFHDS2 as we parse them. Let's see if we can get a definition from FlySky.

pfeerick, 2023-02-15:
So this is mainly an AFHDS3 telemetry parsing issue then? Not that surprising since the AFHDS3 stuff is all pretty new here. :grin:

CharlesGauci, 2023-02-18:
I have the same issue with the FTR10 receiver on the EL18.

CharlesGauci, 2023-02-24:
These errors are similar to the NV14 readings before September 2020, until qba667 made some updates and solved the problem. You can see the history in RCGroups 'Underground FPV Dark Knight Radio' pages 247 to 249. Eshifri used to follow these updates, maybe he can help.

pfeerick, 2023-02-25:
Link to p247 for ease of access: https://www.rcgroups.com/forums/showthread.php?3023832-UndergroundFPV-Dark-Knight-Radio/page247

CharlesGauci, 2023-02-25:
Github changes are mentioned in page 250 by 'yds,',  report number 3741. Sorry, I do not know how to create a link 😓

https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/commits/pulses23?after=517cb176179e691644432db3b8516905b65864ba+69&branch=pulses23&qualified_name=refs%2Fheads%2Fpulses23

PR #3467 (2023-04-19): (v2.8) Fix the bug that some sensor values are displayed incorrectly

For v2.8 branch

Fixes #3196

Summary of changes:
closed 2023-04-19 by pfeerick