EdgeTX/edgetx #3198: Error with LUA script from version of EDGE 2.9.0 onwards

koaxheli, 2023-02-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Lua Script from Mike Shellin Crow-aware Adativ Trim v3.x 
https://rc-soar.com/opentx/lua/adaptivetrim/
was error free until firmware of EDGE 2.8.0.
From v2.9.0 error when reading the curve values.

Compilation with Github.
Code for v2.8
https://gitpod.io/#https://github.com/EdgeTX/edgetx/tree/2.8

Code for v2.9
https://gitpod.io/#https://github.com/EdgeTX/edgetx/tree/main

cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DHELI=YES -DTRANSLATIONS=DE -DBLUETOOTH=YES -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release ../

As from  >= Code 2.8
make configure

make -C arm-none-eabi -j`nproc` firmware

Rewrite firmware.bin to tx16s-xxx.2.9.0.bin
cd arm-none-eabi
mv firmware.bin tx16s-blu-de-gps_2.9.0.bin

### Describe the solution you'd like

Has something been changed with FW 2.9.0.

### Additional context

[temp.zip](https://github.com/EdgeTX/edgetx/files/10756634/temp.zip)

raphaelcoeffic, 2023-02-16:
The issue is the following: we fixed the consistency of the table indexes returned by LUA functions.

Previously, some tables would return `0`-indexed tables, and some would return `1`-indexed tables. However, LUA uses `1`-indexed tables and we decided to improve the compliance to LUA standards and at the same time the consistence of returned results within the EdgeTX LUA API.

@RC-SOAR I think we can work something out for `2.9`. That will probably imply testing the EdgeTX version to know how to adapt to the changing API. Otherwise, you could iterate differently through the table by basically enumerating it rather than accessing indexes directly by using the `pairs()` function:
```lua
for k,v in pairs(t) do
 print(k,v)
end
```

RC-SOAR, 2023-02-16:
@raphaelcoeffic thanks for the heads up - I'll check it out.  Will a 2.9 nightly of Companion be available soon? (I tried to download a nightly from my Companion 2.8, but it failed with 'no assets found in release 'nightly' using filter '^edgetx cpn win64')?

raphaelcoeffic, 2023-02-16:
You should be able use the latest build from `main`: https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain

gagarinlg, 2023-05-06:
@RC-SOAR did you have time to test?

RC-SOAR, 2023-05-06:
Looks like some refactoring is involved and last time I looked it was not easy to debug 2.9. I normally use Companion sim output, if Companion is available as a win executable or there is an (easy) alternative means of debugging then I can have a look next week.

pfeerick, 2023-05-07:
You can always get the latest development build of Companion for Win64 from here: https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain

RC-SOAR, 2023-05-08:
@raphaelcoeffic  Okay so I'm back home and ready to modify the script after the Lua changes... however looking at the Companion build, it's marked with 'failure' status and the output shows a build error. Any chance this can be fixed soon?

gagarinlg, 2023-05-08:
@pfeerick was looking into this build problem. Some MSYS f*up

RC-SOAR, 2023-05-10:
Progress: I've modified the adaptive trim script: calls to model.getCurve have been replaced with calls to a local getCurve() function which tests curve.y[0] and if not nil performs a copy with offset. Code below, I'm certainly open to suggestions for a better/cleaner way.

```
local function getCurve(idx)
  local cv = model.getCurve (idx)
  if cv == nil then return nil end
  
  -- initialise copy
  local cvDest = {}
  cvDest.y = {}
  
  -- Base 0 or 1
  local j = cv.y[0] and 0 or 1
  
  -- copy and shift
  for i = 1, cv.points do
    cvDest.y[i] = cv.y[j]
    j = j + 1
  end
  cvDest.smooth = cv.smooth
  cvDest.points = cv.points
  return cvDest
end
```

Tested okay in OTX Companion simulator. It still needs testing with ETX 2.9 - it would be really useful to have the ETX Companion executable for this.

pfeerick, 2023-05-10:
Indeed. The best I can offer atm until we find out *why* the windows builds spontaneously combusted is a copy of a PR build from  a month ago... http://files.portabledev.xyz/f/4e15cbf215b74a439306/?dl=1

RC-SOAR, 2023-05-10:
Thanks, the download worked fine.

The modded script is now tested in Companion and on my TX16S hardware, using the latest nightly build (downloaded using Buddy). It's working as expected. I'll post the updated script (version 3.2) to the website in due course.

closed 2023-09-20 by pfeerick