EdgeTX/edgetx #3207: main - SD Logging not working

pfeerick, 2023-02-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With a nightly build, SD Logging is not writing to SD card on radio, but is on simulator. Unsure how long this has been the case, but 2.8.1 is functioning correctly.

@mha1 You love the logging stuff... any idea what broke? Affects colorlcd and B&W hardware, but not simulator?

### Expected Behavior

That it write to the SD card

### Steps To Reproduce

1. Set a SF to trigger SD logging on a swtich, 0.5s (interval not important)
2. Let it run a few seconds
3. Stop the SF (move switch to the not-active position)
4. View the content of LOGS folder - no file will be present

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster Boxer, Radiomaster TX16S / TX16SMK2, Radiomaster Zorro

### Anything else?

https://github.com/EdgeTX/edgetx/commit/e930183395fb9d3866eaeeec09752b40d6fe96db

mha1, 2023-02-18:
I'll see if I can find anything

mha1, 2023-02-18:
It stopped working with @raphaelcoeffic's PR https://github.com/EdgeTX/edgetx/pull/3055 (chore: new module/serial API). This PR obviously changed the logic (or timing) for `bool mixerTaskRunning()`. logs.cpp used this to determine if logsWrite() can be called without disturbing the mixer task. I believe this is a leftover from the pre-software timer days. With using the software timer task and having it run at a lower task priority than the mixer task the check for "is the mixer task running" is obsolete as software timers can't interrupt the mixer task. Removing the check brings back writing logs to the SD card.

I have put in PR #3208 to fix this issue.

Interestingly enough there is another software timer (Space Mouse) doing the same. I bet Space Mouse also won't work and also needs to have the  `if (!mixerTaskRunning())` removed. Can you test space mouse?

```
static void spacemouseTimerCb(TimerHandle_t xTimer)
{
  if (!mixerTaskRunning()) {
    DEBUG_TIMER_START(debugTimerSpacemouseWakeup);
    spacemousePoll();
    spacemouseWakeup();
    DEBUG_TIMER_STOP(debugTimerSpacemouseWakeup);
  }
}
```

mha1, 2023-02-18:
Oh, and the simulator wasn't affected as we don't run logs through software timers with SIMU.

closed 2023-02-18 by rotorman

pfeerick, 2023-02-19:
Thanks for tracking this down so quickly :)