EdgeTX/edgetx #3211: Expo Curve does not display correctly

rellik75, 2023-02-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

From the input tab, insert a new line below any input (new aileron line, for example). Edit the new line and configure expo. The curve graph does not update.

### Expected Behavior

The curve graph should update as the expo value is set. 

### Steps To Reproduce

From the input tab, insert a new line below any input (new aileron line, for example). Edit the new line and configure expo. The curve graph does not update. Exit the edit screen and edit the first (original line). Set an expo value and the curve graph will update. Exit the edit screen and go back to the inserted line. The curve graph has now updated. 

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-02-19:
I suspect this is the same issue as #3057

pfeerick, 2023-02-20:
Indeed it sounds like it is. In other words, ETX behaviour, inherited from OTX, is to show the *active* expo/input line, not the current one. but there is a PR proposed to add a toggle option for this. 

Duplicate of #3057

closed 2023-02-20 by pfeerick