EdgeTX/edgetx #3212: Spectrum Analyzer log

Luznatural, 2023-02-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Recently we are having in our club problems related to interferences in 2.4 Ghz that we suspect are due to an inhibitor of signal, as they appear only on weekends and at certain specific hours. Very frequently, we crash several planes at the same time, in separating places. Radios affected are of different class, but all of them in 2,4Ghz: Futaba, Frsky, Radiomaster, etc.
In order to produce evidences for making the opportune denounce to the authorities, I would like to have any capability for recording logs of the Spectrum Analyzer that we could reproduce in the presence of the police.
When you start an inhibitor on, all signals through the spectrum should rise at the same time to very high levels, that could be perfectly identified.

### Describe the solution you'd like

When entering in the Lua script for the Spectrum Analyzer, I would like to have:
1.- A button in the lower part of the screen labeled "Record" to start recording the data. When you press that button, the label could change to say "stop recording" to stop the process. All logs created should contain information on the date and the time they were recorded, when reproduced.
2.- A second button labeled "Reproduce existing logs". When you press that button, the screen should show the list of logs recorded in a folder of the SD Card, and allow you to select any one. Once selected, the log should be reproduced in the screen, in the same format of the regular Spectrum Analyzer, but showing date and time of the recording. Capability to pause it would br welcomed. Ideally, a time scale could be available to move quickly from one part to another of the log, just maintaining the finger on the screen and dragging it from left to right, or from right to left.
3.- A third button labeled "settings" that allows selecting the place in the SD card where the information is to be stored, and/or selected to be reproduced.
4.- Ideally, the files should be reproduced in any radio with Spectrum Analyzer, as far as you place them in the adequate folder.
 

### Describe alternatives you've considered

Any that could be shown to th authorities would be valid

### Additional context

I will add pictures with my suggestion, as soon as I have them available. Thanks in advance.

rueckwaertsflieger, 2023-02-23:
Just asking. Are you sure about the inhibitor being an in-band (2.4GHz) thing? Again just asking. What is an inhibitor? Assuming, you speak of a jammer under suspicion.
If you already crashed several planes...   ... in germany here is a public service from the regulating authority BNetzA, searching for jammer. Any similar service in your country?
PS: When it comes to directional radio, immense field strength is created, causing total receiver blindness, even if signal is out of band.

Luznatural, 2023-02-24:
Thanks for asking. I was fearing nobody was interested on this issue.
An inhibitor is an electronic set that is able to jam one or several frequencies or bands. Normally they are built for using against IED (improvised Explosive Devices) and is very frequent to see some of those in official or military vehicles. They are also used in anti-drone operations, when necessary.
We have also in Spain the service you describe happening in Germany, but you need to make a claim based on facts. Nowadays is very easy to find devices able to jam 2,4 Mhz, especially in the Chinese market, and despite the fact that they are illegal.
Other possibility we are considering is that perhaps a legal company which is producing legal anti-drone systems, is jamming unconsciously towards our activities.
In any case, the Police told us is far better to establish a case, to show any kind of proof to enhace the possibilities to solve the problem.
And that's why we would like to have any recording capability of the Spectrum Analyzer offered by multi module based radios.
One commercial spectrum analyzer normally requires 220volts on the field, that we do not have.