EdgeTX/edgetx #3213: Image size check

gagarinlg, 2023-02-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

loading oversized images can crash the radio

### Describe the solution you'd like

check image size in pixels and in bytes before loading
check model images to be not larger than 1.5x the recommended image sizes and display a warning if a model image is to large

### Additional context

OpentX/Ethos libopenui got that added a short while ago
https://github.com/opentx/libopenui/commit/aedbce438725e003de4c0167051fadf7c34af9d7

pfeerick, 2023-02-20:
I'd take it further, and outright reject the file (at 1.5x recommended size, or perhaps 2x), as some people will always ignore the warning, and then wonder later why stuff is going wrong.

3djc, 2023-02-20:
I'm not even sure that anything larger than screen size should be allowed, it's wasted ram