EdgeTX/edgetx #3214: No Flashing 2.8.1 to Taranis X9DPLUS

ThePhantom05, 2023-02-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Taranis X9D Plus and, accordingly, Taranis X9D, I think too, is not stitched at all, not like! Not through the STM32 loader, not through the companion Edge TX and not through the loader in the transmitter of the above. When flashing in these ways, vibration and a bright white screen appear (while the blue socket is turned on in the transmitter). Pulling out the battery or firmware via STM32 on open tx also helps!

### Expected Behavior

I expected that the problem was fixed, but alas! BEFORE that, it somehow worked and went out in the menu, and this is generally vibration and a bright white screen!!!!

### Steps To Reproduce

1 firmware option via the remote loader.
I threw the firmware through the memory card - the latest release 2.8.1 - connect to the USB bootloader to open the tx - firmware recording - select x9d-2.8.1.bin the recording strip runs - goes back to the bootloader to open tx - I go down and select exit and a long vibration and a bright white screen and so on until you turn off the transmitter! If enabled, the above problem persists! Only the bootloader starts if it is not updated x9d-2.8.1.bin! If you update the firmware only via bootloader STM32! - backup (OPEN TX)
option 2 firmware via companion
You download x9d-2.8.1.bin, that is, you specify in the downloads and select a picture - click the update firmware button on the transmitter - the recording strip runs; after turning on the transmitter - the firmware has not changed - OPEN TX has remained...
The memory card in these 2 cases is loaded as for EDGE TX.
Option 3 firmware via STM32
To be honest, it gave an error for the first time; at first - as far as I understood, since there is little knowledge in the firmware of controllers - by fuses! The second time it was stitched, but alas - the problem is as indicated in the first option (long vibration and a bright white screen) and turning off and removing the battery does not help.!

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+

### Anything else?

The answer to the question
What version of EdgeTX software are you running?
EdgeTX "Flying Dutchman" v2.8.1

pfeerick, 2023-02-20:
` x9d-2.8.1.bin` is not the correct firmware for the Taranis X9D Plus. You should be flashing `x9dp` for the X9Dplus. If you incorrectly flash the x9d firmware onto an x9dplus, yes, it will vibrate crazy and such. Don't ask how I know that. 😇

ThePhantom05, 2023-02-20:
OK) I'll try to do everything else tonight! But on the other hand - I also flashed through the companion 2.8.1 there should be no problems there, he downloaded everything himself. Only the multiprotocol did not update (1.3.7), but I don't think that's because of this! Okay, I'm checking again today! And I'll write after - what happened!!!
Thanks for the quick response!

Luznatural, 2023-02-20:
I updated my X9DP and my X9DP SE 2019, updating the bootloader first, and then via bootloader mode, and everything worked perfectly.
I hope you solve it!!!

ThePhantom05, 2023-02-20:
That's right, I chose the wrong firmware! But if you add your picture to the screensaver through a companion, the transmitter does not turn on after the firmware. 
Or do I need to flash the original first and then add a picture?

ThePhantom05, 2023-02-20:
> Правильно, [pfeerick] Я выбрал не ту прошивку! Но если добавить свою картинку на заставку через компаньона, то после прошивки передатчик не включается. Или нужно сначала прошить оригинал, а потом добавить картинку?

I checked the original was updated first, then I asked for it through a companion with a picture - the problem above remains!

Luznatural, 2023-02-20:
> > That's right, [pfeerick] I chose the wrong firmware! But if you add your picture to the screensaver through a companion, the transmitter does not turn on after the firmware.
> > Or do I need to flash the original first and then add a picture?
> 
> I checked, flashed the original updated, then I flashed through a companion with a picture - the above problem remains!

Try making the size of the picture smaller. If the picture of the screen saver is too big, the screen is not able to present it and collapses.

pfeerick, 2023-02-21:
Most likely what is happening is when you add a custom splash screen, the firmware is now too big.

ThePhantom05, 2023-02-22:
I don't understand! The open tx firmware has 512 kb in memory with a custom picture and ppmus, and the Edge TX firmware without a picture weighs 480 kb - it shows everything in memory built into Taranis (when I connect it via usb via the bootloader) And it works stably....

pfeerick, 2023-02-23:
Very strange indeed... I've now flashed three different firmware images with three different images, and all three display properly. 

As long as you don't get an error when adding the splash image to the firmware, it should work. If you still get the stock logo, you've probably either not edited the firmware file properly, or haven't flashed it. Unless there is an unexpected bug lurking. The firmware doesn't get any bigger... there is space reserved in it for the splash screen - which also dictates the maximum size. 

Can you give these a try and see if they work for you also? Maybe you can send me the image you're using and I can try it on my end? (probably best to zip it, and you can just drag the zip to the text area). And just to be clear/rule it out, you are using EdgeTX Companion to alter the splash screen? As OpenTX Companion won't encode the image properly. 

[x9dp.zip](https://github.com/EdgeTX/edgetx/files/10811511/x9dp.zip)

gagarinlg, 2023-02-23:
Are you using the correct version of Companion? It has to match the radio firmware version.

ThePhantom05, 2023-02-24:
> Вы используете правильную версию Companion? Она должна соответствовать версии прошивки радио.

I used 2.8.0 companion most likely something with it. Updated via the above mentioned version - that's it. I deleted 2.8.0 and downloaded the new 2.8.1. I'm trying with it.

ThePhantom05, 2023-02-24:
> Very strange indeed... I've now flashed three different firmware images with three different images, and all three display properly.
>
> As long as you don't get an error when adding the splash image [...]

My picture is in the archive and the settings you need (ppmus heli lua)[](url) https://github.com/ThePhantom05/Edge-TX-/blob/main/Splash%20Screen.rar  
P,S. During the day I will try to flash according to the above parameters/

ThePhantom05, 2023-02-24:
> Very strange indeed... I've now flashed three different firmware images with three different images, and all three display properly.
>
> As long as you don't get an error when adding the splash image [...]

The screen is stitched and loaded with your picture!
What's wrong with my picture....I don't understand with OPENTX this picture works - I created in Photoshop so that grayscale in 8 or 16 bit mode was saved...

ThePhantom05, 2023-02-24:
> Very strange indeed... I've now flashed three different firmware images with three different images, and all three display properly.
>
> As long as you don't get an error when adding the splash image [...]
Please create a firmware with my picture with the parameters - heli, ppmus, (lua-this is necessarily how I use the custom Music2.Lua).

ThePhantom05, 2023-02-24:
Firmware report via DFU STM32 - with my picture
C:/Program Files/EdgeTX/Companion 2.8/dfu-util.exe -a 0 --dfuse-address 0x08000000 --device 0483:df11 -DC:/Users/~The Phantom~/AppData/Local/Temp/2136-flash.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... ID 0483:df11
Run-time device DFU version 011a
Found DFU: [0483:df11] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 492400
.................................................................................................................................................................................................................................................
File downloaded successfully
No valid DFU suffix signature
Warning: File has no DFU suffix

=================================================================================

ThePhantom05, 2023-02-24:
After edit settings with all the ticks indicated in the screenshot, I press the button and the window indicated in the following screenshot appears...I used to do this with OPEN TX and everything worked! yes, I updated the picture 8 bits, but without success! 
[image] 
[image]

ThePhantom05, 2023-02-24:
It turns out that there is an edit radio splash image in the companion (I never used it) - and then I decided to try it for me when adding an image to the firmware - I click save and gives this error - compressed image size ex ceeds reverse space. 

If I understand correctly, the error is related to space? And what to do?
At the moment, this is the firmware that was sent to me! Weighs 512 kb

ThePhantom05, 2023-02-24:
I attach a jpeg of my screensaver just in case!
[image]

gagarinlg, 2023-02-24:
There is a certain amount of space reserved for the splash screen. When you use the "edit radio splash image" function, the image is injected into the firmware.
With 2.6 or 2.7 we had to reduce the amount of spaced reserverd for the splash image, to make room for new features. To be able to store screen sized images, we had to add a compression for the splash screen. When the compressed image exceeds the available space for the splash screen, this error is given.
Except for reducing the details in your image there is nothing we or you can do.

ThePhantom05, 2023-02-24:
> Определенное количество места зарезервировано для экрана-заставки. Когда вы используете функцию «редактировать изображение заставки радио», изображение вставляется в прошивку. В версии 2.6 или 2.7 нам пришлось уменьшить размер резервирования изображения-заставки, чтобы освободить место для новых функций. Чтобы иметь возможность хранить изображения размером с экран, нам пришлось добавить сжатие для экрана-заставки. Когда сжатое изображение превышает доступное пространство для экрана-заставки, выдается эта ошибка. Кроме уменьшения деталей вашего изображения, мы или вы ничего не можете сделать.

Reduced the size to 6.5 kilobytes, but still the error above is indicated. Reduce the size of the details of the picture?
Or tell me the maximum size for the splash screen...how much should it be?

ThePhantom05, 2023-02-24:
> There is a certain amount of space reserved for the splash screen. When you use the "edit radio splash image" function, the image is injected into the firmware. With 2.6 or 2.7 we had to reduce the [...]

Reduced the size to 6.5 kilobytes, but still the error above is indicated. Reduce the size of the details of the picture?
Or tell me the maximum size for the splash screen...how much should it be?

ThePhantom05, 2023-02-24:
Thanks) For the information provided! Everything worked out! You can close I think the problem!

ThePhantom05, 2023-02-24:
Dear pfeerick, please create a firmware for with heli parameters, ppmus lua for x9dplus. Through the companion does not come out, add them. The picture as I wrote everything turned out) There is no heli menu in my transmitter.

rotorman, 2023-02-24:
@ThePhantom05 have a look at: https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod
For pre-2019 X9D+ with heli and PPM unit of µS from v2.8.1 tag you would use:
https://gitpod.io/#https://github.com/edgetx/edgetx/tree/v2.8.1
```bash
cmake -Wno-dev -DCMAKE_RULE_MESSAGES=OFF -DPCB=X9D+ -DHELI=YES -DPPM_UNIT=US -DCMAKE_BUILD_TYPE=Release ../
make arm-none-eabi-configure
make -C arm-none-eabi -j`nproc` firmware

ThePhantom05, 2023-02-24:
Thanks you) Ok)

closed 2023-02-24 by ThePhantom05