EdgeTX/edgetx #3220: UI 2.9 Rework Status

JimB40, 2023-02-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This issue is for UI rework status overview
_Last update: 2023-02-21_

**PAGES**

- RADIO SETTINGS
  - TOOLS 
    - TOOLS (_main page_) - 2.8 version no change
    - Spectrum Analyser (_sub page_) - https://github.com/EdgeTX/edgetx/pull/3077
    - GHOST MENU (_sub page_) - 2.8 version no change

  - SD-HC CARD
    - SD-HC CARD (_main page_) - 2.8 version no change

  - RADIO SETUP 
    - RADIO SETUP (_main page_) - 2.8 version no change
    - Sound (_sub page_) - 2.8 version no change
    - Variometer (_sub page_) - 2.8 version no change
    - Haptic (_sub page_) - 2.8 version no change
    - Alarms (_sub page_) - 2.8 version no change
    - Backlight (_sub page_) - 2.8 version no change
    - GPS (_sub page_) - 2.8 version no change

  - THEMES
    - THEMES (_main page) - 2.8 version no change
    - EDIT THEME (_Theme edit page_) https://github.com/EdgeTX/edgetx/pull/3078
    - Edit Color (_Color selection page_)  - 2.8 version no change

  - GLOBAL FUNCTIONS 
    - GLOBAL FUNCTIONS (_main page, list_) - https://github.com/EdgeTX/edgetx/pull/2992
    - GLOBAL FUNCTIONS (_Global functions edit page_) - https://github.com/EdgeTX/edgetx/pull/2992

  - TRAINER
    - TRAINER (_main page_) - https://github.com/EdgeTX/edgetx/pull/3104

  - HARDWARE
    - HARDWARE (_main page_) - 2.8 version no change
    - CALIBRATION (_sub page_) - 2.8 version no change
    - ANALOGS/Calibrated Analogs (_sub page_) - 2.8 version no change
    - ANALOGS/Filtered raw analogs with deviation (_sub page_) - 2.8 version no change
    - ANALOGS/Unfiltered raw analogs (_sub page_) - 2.8 version no change
    - ANALOGS/Min, max. and range (_sub page_) - 2.8 version no change
    - KEYS/SWITCHES TEST (_sub page_)  - 2.8 version no change

  - VERSION
    - VERSION (_main page_)  - 2.8 version no change

- MODEL SETTINGS
  - MODEL SETUP
    - MODEL SETUP (_main page_) - 2.8 version no change
    - Internal RF (_sub page_) - 2.8 version no change
    - External RF (_sub page_) - 2.8 version no change
    - Trainer(_sub page_) - 2.8 version no change
    - Timer 1-3 (_sub page_) - 2.8 version no change
    - Preflight Checks (_sub page_) - 2.8 version no change
    - Trims (_sub page_) - 2.8 version no change
    - Throttle (_sub page_) - 2.8 version no change

  - HELI SETUP
    - HELI SETUP (_main page_) - 2.8 version no change
 
  - FLIGHT MODES
    - FLIGHT MODES (_main page, list_) - 2.8 version no change
    - FLIGHT MODE 0-8 (_Flight mode edit page_) - 2.8 version no change
    
   - INPUTS
     - INPUTS (_main page, list_) - 2.8 version no change
     - INPUTS (_Input edit page_) https://github.com/EdgeTX/edgetx/pull/2939
 
   - MIXES 
     - MIXES (_main page, list_) - 2.8 version no change
     -  MIXES (_Mixes edit page1_) - 2.8 version no change
     -  MIXES (_Mixes edit page2_) - 2.8 version no change
  
  - OUTPUTS
    - OUTPUTS (_main page, list_)- 2.8 version no change
    - _no header_ (_Output edit page_) - 2.8 version no change

  - CURVES
    - CURVES (_main page, list_) - https://github.com/EdgeTX/edgetx/pull/3041
    - CURVE (_Curve edit page_) - https://github.com/EdgeTX/edgetx/pull/2953

  - GLOBAL VARIABLES
    - GLOBAL VARIABLES (_main page, list_) https://github.com/EdgeTX/edgetx/pull/2945
    -  Global Variable  (_Global Variable edit page_) - ???

  - LOGICAL SWITCHES
    - LOGICAL SWITCHES (_main_page, list_) https://github.com/EdgeTX/edgetx/pull/3019
    -  LOGICAL SWITCHES  (_Logical Switches edit page_) - ???

  - SPECIAL FUNCTIONS
    - SPECIAL FUNCTIONS (_main_page, list_) https://github.com/EdgeTX/edgetx/pull/2992
    - SPECIAL FUNCTIONS (_Special function edit page_) https://github.com/EdgeTX/edgetx/pull/2992

  - CUSTOM SCRIPTS
    - CUSTOM SCRIPTS (_main_page, list_) https://github.com/EdgeTX/edgetx/pull/3071
    - CUSTOM SCRIPTS (_Custom scripts edit page_) https://github.com/EdgeTX/edgetx/pull/3071

  - TELEMETRY
    - TELEMETRY (_main page, list_) https://github.com/EdgeTX/edgetx/pull/3070
    - SENSOR (_telemetry sensor edit page_) https://github.com/EdgeTX/edgetx/pull/3070
  
- MODEL SELECT
   - Select Model (_Model browser_)  - 2.8 version no change
   - Select a template folder (_New Model creation_) https://github.com/EdgeTX/edgetx/pull/3076

- CHANNEL MONITOR
  - CHANNELS MONITOR 1-8 (_sub page_) - 2.8 version no change
  - CHANNELS MONITOR 9-16 (_sub page_) - 2.8 version no change
  - CHANNELS MONITOR 17-25 (_sub page_) - 2.8 version no change
  - CHANNELS MONITOR 25-32 (_sub page_) - 2.8 version no change
  - LOGICAL SWITCHES MONITOR 1-8 (_sub page_) https://github.com/EdgeTX/edgetx/pull/3019

- SCREEN SETTINGS
  - User interface (_sub page_) - 2.8 version no change
  - Main View 1-5 (_sub page_) - 2.8 version no change

- STATISTICS 
  - Statictics (_sub page_) - https://github.com/EdgeTX/edgetx/pull/3072
  - Debug (_sub page_) - https://github.com/EdgeTX/edgetx/pull/3072
  
**MODALS (_Pop-up windows)**
- TEXT INPUT (keyboard)
- NUMBER INPUT
- LIST
- LIST FILTERED (source selection)
- LIST FILTERED (file selection
  - IMAGE selection
  - Sound selection
  - LUA selection

**UI ATOMS & STYLING**
- Text
- Text with icon
- Button with text
- Button with  icon
- Button with text & icon
- Button for list selection
- Check-box
- Switch
- Switch with text
- Slider
- List line
- List line with sub-lines    

### Describe the solution you'd like

-

philmoz, 2023-02-21:
Logical switches edit page was converted to lvgl in #3019 
Global variables list was changed to match other pages in #3116 
Global variables edit page was converted to lvgl in #3116

jawz101, 2025-01-22:
Instead of user themes, why not simply have a light mode and a dark mode with changeable accent color?  All of the themes are fugly.

pfeerick, 2025-01-23:
With themes, that is a completely unnecessary restriction / removal of capabilities, isn't it? There is absolutely nothing to stop you creating light and dark modes via themes, as these are really just themes themselves, if the capabilities of the theming system are sufficiently capable.