EdgeTX/edgetx #3221: UI. 2.9 Text field styling

JimB40, 2023-02-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Current styling
[image]

### Describe the solution you'd like

New styling:
- frame height 32px
- number vertically centered
- number horizontally justified to the right with minimum 4px padding 
[image]

philmoz, 2023-04-15:
I've added PR #89 to the libopenui project to implement this.
It also adjusts the text input control size as well.

[image]

gagarinlg, 2025-01-24:
I would say this should be part of the 3.0 UI rework